ros/ros_comm #1900: AttributeError: 'NoneType' object has no attribute 'buff_size'

khssnv, 2020-02-26:
# Summary
Exception occurs after `rospy.wait_for_message()` call returns value (invoked in HTTP request handler).

# Setup information
ROS version: melodic
OS version: Ubuntu 18.04
Python version: python3.6

# More details
I make HTTP endpoint returns latest message from topic. There is a `Publisher` created with `latch` parameter set and subscriber use `rospy.wait_for_message()` to get message into HTTP endpoint handler. Please find an example below. As you may find in logs, `wait_for_message` returns message and HTTP endpoint handler returns it to client. But there are exceptions from `tcpros_base` occurs after `wait_for_message` call.

# Minimal example

## Web server "Alice"
Provides `/bob/pub` HTTP GET endpoint. Contains `wait_for_message` call. Provides exceptions mentioned.

<details>
  <summary>alice.py</summary>

```python
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

import uvicorn
from fastapi import FastAPI
from starlette.responses import Response

import rospy
from std_msgs.msg import String

app = FastAPI()

@app.get("/bob/pub")
def handler():
    msg = rospy.wait_for_message("/bob/pub", String, timeout=1)
    return msg.data

if __name__ == "__main__":
    rospy.init_node("alice", log_level=rospy.DEBUG)
    uvicorn.run("alice:app", host="localhost", port=8081, log_level="info")
```

</details>

## Publisher "Bob"

<details>
  <summary>bob.py</summary>

```python
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

import rospy
from std_msgs.msg import String

rospy.init_node("bob", log_level=rospy.DEBUG)
pub = rospy.Publisher("~pub", String, queue_size=10, latch=True)
def make_pub(event):
    pub.publish(String(data="bob text"))
rospy.Timer(rospy.Duration(secs=1), make_pub)
rospy.spin()
```

</details>

# Execution logs
You could find exceptions mentioned in `alice.py` logs.

<details>
[... 165 lines cut ...]

</details>

<details>
  <summary>$ rosrun bob bob.py</summary>

```bash
$ rosrun bob bob
[DEBUG] [1582726911.809096]: init_node, name[/bob], pid[32357]
[DEBUG] [1582726911.811201]: binding to 0.0.0.0 0
[DEBUG] [1582726911.812757]: bound to 0.0.0.0 36993
[DEBUG] [1582726911.814646]: ... service URL is rosrpc://32vd:36993
[DEBUG] [1582726911.816123]: [/bob/get_loggers]: new Service instance
[DEBUG] [1582726911.819473]: ... service URL is rosrpc://32vd:36993
[DEBUG] [1582726911.821454]: [/bob/set_logger_level]: new Service instance
[DEBUG] [1582726911.826141]: node[/bob, http://32vd:33177/] entering spin(), pid[32357]
[WARN] [1582726917.061248]: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[WARN] [1582726920.083261]: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[WARN] [1582726920.956348]: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[WARN] [1582726921.901228]: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[WARN] [1582726922.832478]: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[WARN] [1582726924.450617]: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
```

</details>

dirk-thomas, 2020-02-28:
It looks like a race condition in which the socket has been closed and therefore various variables have been reset to `None`. Please consider to contribute a pull request targeting the default branch (currently `noetic-devel`).

Privilger, 2020-05-08:
I also meet with this problem. My case is that I speeded up the Gazebo and then after around 10mins, this bug raised.

jikawa-az, 2021-02-24:
Hello, we have a customer encountering this issue, did you ever find a solution or workaround?
Thank you!

anmakon, 2023-01-31:
I encounter the same bug with my robot, unfortunately still running on ROS kinetic though. Would anyone have an idea if this could be fixed in ROS noetic by now?

closed 2025-05-31 by sloretz