ros/ros_comm #1901: Inconsistent message queue handling between roscpp and rospy

sfalexrog, 2020-02-27:
(Started at https://github.com/CopterExpress/clever/pull/218)

## The problem

While writing some image processing code in Python, we have noticed that if the code is being run inside the callback and is taking a while to complete, the latency would increase significantly over time. This happens even with `queue_size=1` set in publisher and subscriber, and is only partially affected by `buff_size`.

In order to reproduce the problem, I did the following:
* wrote a simple node that subscribed to a message, looked at the difference between `rospy.Time.now()` and the message timestamp, and blocked the callback thread for one second (using `rospy.sleep()`);
* wrote the same code in C++ (I've blocked the thread by using `std::this_thread::sleep_for`);
* looked at the resulting differences.

Typical message "staleness" (the difference between the moment when the callback was called and the message timestamp) was close to the time between two published messages for the C++ variant; in Python, however, it was closer to the time it took for the callback to finish.

## Test setup

I've tried to write a simple test to determine the staleness for different setups; the code is available here: https://github.com/sfalexrog/staleness_test. I also tried running the code in Travis: https://travis-ci.com/sfalexrog/staleness_test/.

There are three variants for publishing nodes (`nodelet` - the node is a C++ nodelet; `node_cpp` - the node is a C++ node; `node_py` - the node is a Python node) and three variants for subscribing nodes (which have the same names). Additionally, there is a variant of the `node_py` node, called a `node_py_largebuf`, which is basically the same as `node_py`, but with `buff_size` set to approximately the size of messages generated by the publisher during the callback.

Publishers simulate a camera that generates 640x480 RGB8 images at 100 Hz. Subsribers simulate image processing nodes with long-running code (the message callback blocks for 1 second). Publishing and subscribing nodelets use the same nodelet manager.

## Test results

Below are the average results from running on Kinetic (https://travis-ci.com/sfalexrog/staleness_test/jobs/291735129); column names describe the publisher type, row names describe the subscriber type:

|   |`nodelet`|`node_cpp`|`node_py`|
|---|---|---|---|
|`nodelet`|0.000253|0.005502|0.006024|
|`node_cpp`|0.005065|0.00517571|0.005521|
|`node_py`|**8.126688**|**8.159472**|**8.077343**|
|`node_py_largebuf`|**0.944017**|**0.944568**|**0.938216**|

Melodic results are similar: https://travis-ci.com/sfalexrog/staleness_test/jobs/291735130

## Questions

Is this behavior expected or documented? I could not find a warning about long-running subscribers in `rospy`, and the behavior is inconsistent with `roscpp`. We *can* work around this problem by creating a separate thread for long-running tasks, but this makes the code more complicated, and we'd prefer a simplier, more reliable Python API.

mikepurvis, 2020-02-27:
Unfortunately, there are a lot of warts about the rospy implementation, in particular the baffling thread model and lack of user control over what is scheduled and where.

This is not particularly well documented and is very unlikely to change at this point; the path forward on it is to migrate to ROS 2, which is built as a common client library exposing a C interface for which there are bindings that form the basis of each client library:

- https://github.com/ros2/rcl
- https://github.com/ros2/rclcpp
- https://github.com/ros2/rclpy

This design approach is in part in response to these criticisms of ROS 1. It also makes supporting a new language easier in terms of getting broad API coverage right out of the gate.

sfalexrog, 2020-02-27:
@mikepurvis,
Thank you for your suggestion. Unfortunately, there is still a number of packages not readily available for ROS2 (although `ros1_bridge` might probably help with that). The `rclpy` API is also more explicit, which is a mixed blessing - for instance, service calls no longer look like function calls, results are passed as futures (there's a synchronous `call`, but looks like it may be broken in Eloquent), and you can't create a Request implicitly. In my opinion ROS2 is not as mature for educational purposes as ROS1 is (although I would be happy to be proven wrong!).

As for this issue, I believe @okalachev has suggested some ways to fix this `rospy` behavior in the original issue: https://github.com/CopterExpress/clever/pull/218#issuecomment-591170233, https://github.com/CopterExpress/clever/pull/218#issuecomment-591172592. Is there a chance something like that could be done for `rospy`?

mikepurvis, 2020-02-27:
Looks like the suggestion is to create another internal thread for each subscriber exclusively dedicated to reading off the socket, so that doing that is never blocked on long callbacks and the kernel buffer never fills up resulting in loss of the newest data.

This does seem pretty reasonable, and doesn't change any of the semantics of the rospy API. It would be unlikely to break any user code.

If it were possible to implement this without a major overhaul of rospy and could be ready in the next few weeks (eg ahead of the release of Noetic), I would consider it. @dirk-thomas, any further thoughts?

slavanap, 2021-02-10:
Is it possible to just increase socket system buffer for reading?

c-andy-martin, 2021-07-14:
I ran into this problem myself and have coded up a fix, see PR #2170

closed 2025-05-31 by sloretz