ros/ros_comm #1913: Unable to report rosout: __exit__

mm318, 2020-03-19:
One of our ROS applications is seeing the following stack trace:

```
[rospy.rosout][ERROR] 2020-03-18 22:57:59,303: Unable to report rosout: __exit__
Traceback (most recent call last):
  File "/home/robomaker/workspace/applications/robot-application/dependencies/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/rosout.py", line 91, in _rosout
    disable_topics_ = rospy.get_param_cached("/rosout_disable_topics_generation", False)
  File "/home/robomaker/workspace/applications/robot-application/dependencies/opt/ros/melodic/lib/python2.7/dist-packages/rospy/client.py", line 490, in get_param_cached
    return _param_server.get_param_cached(param_name)
  File "/home/robomaker/workspace/applications/robot-application/dependencies/opt/ros/melodic/lib/python2.7/dist-packages/rospy/msproxy.py", line 162, in get_param_cached
    with self._lock:
AttributeError: __exit__
```

It looks like this [change](https://github.com/ros/ros_comm/commit/902fb00fc7b2e881575a270a99c3a077ba5cdbba#diff-b73791b3a9977286c0a4a36d035c4940L90) removed `self._lock = Lock()` and had the intention to remove all other references to `self._lock`, but as of today there is still a reference to `self._lock`:
https://github.com/ros/ros_comm/blob/12eaf946597739e833b2300ef28bb079f248a9e1/clients/rospy/src/rospy/msproxy.py#L162

dirk-thomas, 2020-03-19:
Thanks for the report. Can you please create a pull request removing this left over lock.

dirk-thomas, 2020-03-19:
@mgrrx FYI

closed 2020-03-19 by dirk-thomas