ros/ros_comm #1919: Unknown error initiating TCP/IP socket error ros-kinetic with px4

ashirsat, 2020-03-27:
Hi,

I am trying to run the px4 SITL with ros-kinetic on ubuntu 16.04.

I know that ros-kinetic is end of life and so is Ubuntu 16.04 But i think this issue is with base ros package. Hence I am posting this.

I get this error when I try to roslaunch px4 mavros_posix_sitl.launch file.
PX4 firmware branch is tags/v1.8.0 

The error reported is

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 557, in connect
    self.read_header()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 650, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
AttributeError: 'NoneType' object has no attribute 'buff_size'

Please let me know if there is a workable solution since I am stuck with ros kinetic for a project.

Any help is appreciated.

Thanks,
Aniket

ashirsat, 2020-03-27:
Sorry about the misinformation.

Ubuntu version 16.04 LTS
PX4 Firmware branch: tags/v1.8.0
Python: 2.7

Here are the steps I use.

1) For building px4 firmware sitl: 
I follow their guide for tags/v1.8.0 for ros kinetic listed in the link
https://dev.px4.io/v1.8.0/en/simulation/ros_interface.html
 2) Then i run roslaunch px4 mavros_posix_sitl.launch.
The file causes gazebo7 to start with the default robot sdf file. but it does not complete the process. 
 I wasnte to check out the issue why it is not able to finish connecting the fcu_url in  in the launch file in the link below.

One of the errors I get in the log files is this

[rospy.internal][WARNING] 2020-03-26 17:42:33,412: Unknown error initiating TCP/IP socket to aniketrs-Precision-5510:49955 (http://aniketrs-Precision-5510:35541/): Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 557, in connect
    self.read_header()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 650, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
AttributeError: 'NoneType' object has no attribute 'buff_size'

[rospy.topics][ERROR] 2020-03-26 17:42:33,412: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 326, in close
    c.close()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 838, in close
    self.socket.close()
AttributeError: 'NoneType' object has no attribute 'close'

[rospy.impl.masterslave][INFO] 2020-03-26 17:42:33,412: atexit

I am guessing this error has relevance to why it is not able to connect. 

Also no modification has been made to any file after cloning from the github repo for px4.

@dirk-thomas Let me know if this helps. 

Thanks,
Aniket
https://github.com/PX4/Firmware/blob/v1.8.0/launch/mavros_posix_sitl.launch

dirk-thomas, 2020-03-27:
Please see https://answers.ros.org/question/296544/gazebo-is-aborting/ which seems to describe the same / a similar problem.

It is unlikely that I will have time to debug this. Maybe someone else can with the provided information.

ashirsat, 2020-03-27:
Hey @dirk-thomas,

I agree it is a similar issue. But no one has been able to post a possible work around.
I will try and see if the only answer posted there is of any help and update the response accordingly.

Thanks for quick response.

tfoote, 2020-03-27:
>I agree it is a similar issue. But no one has been able to post a possible work around. 

We can't find a workaround until we've found the root cause. 

With the larger context this looks like a warning that the socket interactions failed. Which is most likely caused by gazebo crashing during the socket operations which is why the socket operations terminate mid flight and cause an exception. However it appears that that error is caught. 

Where does the second error message come from? and the third? Context is important for understanding error messages. To understand what the problem is for the whole system a more complete summary of logs and results will be necessary. The traceback only tells you what part was unrecoverable not what caused it.

dirk-thomas, 2020-04-20:
Closing due to no further reply.

Please feel free to comment with the requested information and the ticket can be reopened.

closed 2020-04-20 by dirk-thomas