ros/ros_comm #1921: rospy logs are not behaving as expected

notenail, 2020-03-28:
Guys, I have been trying to log messages from different topics after a bit of preprocessing. However, when I am examining the file, while it does the job of logging certain topic metadata, it is filled with error messages as shown below. I am relatively new to ROS. Can anyone interpret the message or at least suggest a way to shun it?

`[rosout][INFO] 2020-03-28 17:37:55,449: cam_log, 6429, fb70bb29-1fe4-4154-b077-4eb5a1925d24, 1585397275403970003
[rospy.rosout][ERROR] 2020-03-28 17:37:55,767: Unable to report rosout: __enter__
Traceback (most recent call last):
  File "/home/panick/ros_ws/install/lib/python3/dist-packages/rospy/msproxy.py", line 159, in get_param_cached
    return rospy.impl.paramserver.get_param_server_cache().get(resolved_key)
  File "/home/panick/ros_ws/install/lib/python3/dist-packages/rospy/impl/paramserver.py", line 167, in get
    raise KeyError(key)
KeyError: '/rosout_disable_topics_generation'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/panick/ros_ws/install/lib/python3/dist-packages/rospy/impl/rosout.py", line 91, in _rosout
    disable_topics_ = rospy.get_param_cached("/rosout_disable_topics_generation", False)
  File "/home/panick/ros_ws/install/lib/python3/dist-packages/rospy/client.py", line 490, in get_param_cached
    return _param_server.get_param_cached(param_name)
  File "/home/panick/ros_ws/install/lib/python3/dist-packages/rospy/msproxy.py", line 162, in get_param_cached
    with self._lock:
AttributeError: __enter__`

dirk-thomas, 2020-03-30:
This looks like a duplicate of #1913. Which version of `rospy` are you using? In the latest release (1.15.4) the referenced issue has been resolved.

dirk-thomas, 2020-04-20:
Closing due to no further response.

closed 2020-04-20 by dirk-thomas