ros/ros_comm #1939: protect drop_signal_ with drop_mutex_

fujitatomoya, 2020-04-27:
### Problem Observed

- stack trace

```
gsignal
_gnu_cxx::_verbose_terminate_handler()
std::rethrow_exception(std::__exception_ptr::exception_ptr)
std::terminate()
__cxa_throw
void boost::throw_exception<boost::bad_weak_ptr>(boost::bad_weak_ptr const&)
ros::TransportSubscriberLink::onConnectionDropped(boost::shared_ptr<ros::Connection> const&)
boost::signals2::detail::signal_impl<void (boost::shared_ptr<ros::Connection> const&, ros::Connection::DropReason), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (boost::shared_ptr<ros::Connection> const&, ros::Connection::DropReason)>, boost::function<void (boost::signals2::connection const&, boost::shared_ptr<ros::Connection> const&, ros::Connection::DropReason)>, boost::signals2::mutex>::operator()(boost::shared_ptr<ros::Connection> const&, ros::Connection::DropReason)
ros::Connection::drop(ros::Connection::DropReason)
ros::TransportUDSStream::close()
ros::TransportUDSStream::socketUpdate(int)
ros::PollSet::update(int)
ros::PollManager::threadFunc()
```

### Code Analysis

https://github.com/ros/ros_comm/blob/d82f3f09f2b9b55cbbee68ba25f36174ec2c7002/clients/roscpp/src/libros/transport_subscriber_link.cpp#L122-L135

there is a possibility that TransportSubscriberLink could be deleted already via Publication::dropAllConnections.

https://github.com/ros/ros_comm/blob/d82f3f09f2b9b55cbbee68ba25f36174ec2c7002/clients/roscpp/src/libros/publication.cpp#L306-L323

### Suggested Fix

- Protect drop_signal_ with drop_mutex_, so if TransportLink is already deleted, drop_signal_ is also disconnected.

Barry-Xu-2018, 2020-04-27:
@fujitatomoya 

I submitted a commit to fix this issue. #1940

fujitatomoya, 2020-05-15:
@dirk-thomas @Barry-Xu-2018 

closing this, since PR https://github.com/ros/ros_comm/pull/1940 is merged.

closed 2020-05-15 by fujitatomoya