ros/ros_comm #1946: [rostopic] pub from file not working for messages with arrays

DangitBen, 2020-05-01:
## Expected operation for messages
### Terminal 1:
```bash
roscore
```

### Terminal 2:
```bash
rostopic pub -r 4 test_topic std_msgs/String "data: 'hello'"
```

### Termial 3:
```bash
rostopic echo -n 1 /test_topic > test_topic.bagy
rostopic pub -f test_topic.bagy -r 4 /test_topic std_msgs/String
# Works!
```

## For messages with arrays:
### Terminal 1:
```bash
roscore
```

### Terminal 2:
```bash
rostopic pub -r 4 test_topic2 geometry_msgs/Polygon "points:
- x: 0.0
  y: 0.0
  z: 0.0"
```

### Termial 3:
```bash
rostopic echo -n 1 /test_topic2 > test_topic2.bagy
rostopic pub -f test_topic2.bagy -r 4 /test_topic2 geometry_msgs/Polygon
# Fails
```

Error:
```bash
ERROR: Field [points] must be a list or tuple instead of: NoneType

Args are: [points]
```

Manually copying the contents of the echo'd file and inserting that into the `rostopic pub` command works.

DangitBen, 2020-05-06:
Seems as though switching the output rate flag for a latched topic flag does not cause the crash.

Example:
```bash
rostopic pub -f test_topic2.bagy -l /test_topic2 geometry_msgs/Polygon
```

fujitatomoya, 2020-05-07:
@DangitBen 

so far i got the following result with melodic release,

```
# rostopic pub -f test_topic.bagy -r 4 /test_topic std_msgs/String
publishing and latching message for 3.0 seconds
publishing and latching message for 3.0 seconds
ERROR: Unable to publish message. One of the fields has an incorrect type:
  field data must be of type str

msg file:
string data
```

```
# rostopic pub -f test_topic2.bagy -r 4 /test_topic2 geometry_msgs/Polygon
publishing and latching message for 3.0 seconds
ERROR: Field [points] must be a list or tuple instead of: NoneType

Args are: [points]
```

Actually both failed but publish just once.

DangitBen, 2020-05-07:
@fujitatomoya 
> Actually both failed but publish just once.

I was about to say that I did not see the same behavior, but I was testing w/o anything listening to that topic. When there is a subscriber I see that same error.

I realize I should have put more environment info in my original issue. I too am running Ubuntu 18.04 with melodic release.

I have also found that piping the contents of the file seems to work, but the rate flag is still not working as expected (publishing only once, and quickly exits rather than continually publishing at the specified rate).
```bash
cat test_topic.bagy | rostopic pub -r 4 /test_topic std_msgs/String
```

dirk-thomas, 2020-05-13:
Please consider to contribute a pull request to address this issue.

closed 2025-05-31 by sloretz