ros/ros_comm #1949: assertion fail happens via boost::recursive_mutex when Connection drops.

fujitatomoya, 2020-05-11:
## Core Stack

```
gsignal+0x38/0x78 (libc-2.23.so +0x31528)
uselocale+0x2f4/0x68 (libc-2.23.so +0x2ac04)
__assert_fail+0x4c/0x4c (libc-2.23.so +0x2acac)
_ZN5boost15recursive_mutex4lockEv+0x78/0xa0
_ZN5boost11unique_lockINS_15recursive_mutexEE4lockEv+0x13c/0x230
_ZN3ros10Connection4dropENS0_10DropReasonE+0x84/0x2b4 (libroscpp.so +0xa1554)
```

### Crash Site
https://github.com/ros/ros_comm/blob/c7f7598e8a49b1b4eb6f3e42f698147e151a811d/clients/roscpp/src/libros/connection.cpp#L330

scoped_lock is using typedef boost::unique_lock, assertion with unique_lock is either no mutex or already owns the mutex in the same thread. in this case, expecting the former, which means Connection object owns this drop_mutex_ is gone at this time.

### Expected Scenario

1. Publisher::Impl::unadvertise (thread-A) Publication::dropAllConnections() here it will call ConnectionManager::onConnectionDropped to push the connection in disconnected list. during deleting TransportSubscriberLink list, it gets context switch. (TransportSubscriberLink also possess the shared pointer to connection.)
1. PollManager::threadFunc (thread-B) ConnectionManager::removeDroppedConnections() disconnected list is swapped, so deleting the shared pointers which belonged to ConnectionManager. (at this point, connection objects are still alive, because TransportSubscriberLink still has reference.)
1. PollManager::threadFunc (thread-B) Connection::drop(), at the entrance it gets context switched.
1. thread-A resumes, delete Connection::drop completely. this means reference to connection from TransportSubscriberLink is now gone. Connection object gone.
1. thread-B resumes, accessing Connection object members which are already freed. boom!!!!

Barry-Xu-2018, 2020-05-11:
Use the following sequences could cause an issue that use a destroyed connection object.
 
```
# thread A
 
Publisher::Impl::unadvertise
    TopicManager::instance()->unadvertise
        Publication::drop
            Publication::dropAllConnections
                TransportSubscriberLink::drop
                    Connection::drop
                        drop_signal_
                            ConnectionManager::onConnectionDropped           # step1: push conn into dropped_connections_
 
                        transport_->close();                                 # this operation might be called after step3
 
                V_SubscriberLink local_publishers are end of lifecycle       # step4: erase conn from subscriberlink
 
 
# thread B
 
PollManager::threadFunc
    poll_signal_
        ConnectionManager::removeDroppedConnections                          # step2: erase conn from dropped_connections_
 
    PollSet::update
        TransportTCP::socketUpdate
            Connection::onDisconnect
                Connection::drop                                             # step3
                      boost::recursive_mutex::scoped_lock lock(drop_mutex_); # step5 (boom!)
```
 
Ideal to solve this issue:
 
1. **transport_->close** calls **poll_set_->delSocket** ( PollSet::update deal with the events by filtering socket_fd with exist socket_info_ )
2. connection in ConnectionManager will be released in ConnectionManager::removeDroppedConnections after ConnectionManager::onConnectionDropped is called.
3. to call **transport_->close** before **drop_signal_**, it will make sure that PollSet will not deal with close event of this transport after connection object is destroyed.

venkisagunner93, 2020-05-12:
@fujitatomoya & @Barry-Xu-2018 I'm experiencing a pthread_recursive_mutex error across my codebase. I have publishers and subscribers in a class and I delete and recreate object for that class. Could it be related to this bug ? Also, this only happened to me after I updated my ROS binaries. The following is the back trace. Just commenting to get some clarity on the bug I have and this issue looked pretty close to the issue I was having.

```
/usr/include/boost/thread/pthread/recursive_mutex.hpp:113: void boost::recursive_mutex::lock(): Assertion `!pthread_mutex_lock(&m)' failed

backtrace
#0  0x00007fffeebaefff in raise () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007fffeebb042a in abort () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x00007fffeeba7e67 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#3  0x00007fffeeba7f12 in __assert_fail () from /lib/x86_64-linux-gnu/libc.so.6
#4  0x000055555566120d in boost::recursive_mutex::lock() ()
#5  0x000055555566fce5 in boost::unique_lock<boost::recursive_mutex>::lock() ()
#6  0x00007ffff2ed1795 in ros::Connection::drop(ros::Connection::DropReason) () from /opt/ros/melodic/lib/libroscpp.so
#7  0x00007ffff2f4ca97 in ros::TransportTCP::close() () from /opt/ros/melodic/lib/libroscpp.so
#8  0x00007ffff2f4daa5 in ros::TransportTCP::socketUpdate(int) () from /opt/ros/melodic/lib/libroscpp.so
#9  0x00007ffff2f8b39e in ros::PollSet::update(int) () from /opt/ros/melodic/lib/libroscpp.so
#10 0x00007ffff2f0ae75 in ros::PollManager::threadFunc() () from /opt/ros/melodic/lib/libroscpp.so
#11 0x00007ffff0206116 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.62.0
#12 0x00007fffef9c44a4 in start_thread () from /lib/x86_64-linux-gnu/libpthread.so.0
#13 0x00007fffeec64d0f in clone () from /lib/x86_64-linux-gnu/libc.so.6
```

fujitatomoya, 2020-05-13:
@venkisagunner93 

as far as i see the back trace that you have, it is the same exact one that we have. can you reproduce the issue? if possible, could you import the fix and see if the problem not observed? we are now under evaluation but in our use case, reproduction rate is really low. Or i can do that instead, if you could provide the test sample for reproduction.

venkisagunner93, 2020-05-13:
How can I import the fix. I can test it. I see it every time I run the codebase. But I can't figure out how to reproduce the steps. I even posted in ROS answers and I'm trying to resolve this fix as this stopping other developments unfortunately. If you can guide me how to test this package, I can give it a go. I tried cloning the package and compile and install it. Didn't work. Looks like there are several dependencies that I have to take care of. I can definitely give it a shot, if you can guide me how to import and install this fix. Thanks for replying anyways !

fujitatomoya, 2020-05-13:
> How can I import the fix ?

1. setup catkin workspace for your ROS version. reference here: http://wiki.ros.org/catkin/Tutorials
2. git clone ros_comm into your working and apply patch https://github.com/ros/ros_comm/pull/1950/files
3. catkin build and source your workspace setup.bash
4. reproduce the issue see if the assertion fail still happens.

Barry-Xu-2018, 2020-05-13:
@venkisagunner93 

Based on your backtrace, ROS version used by you is `melodic`.  
#1950 is based on version `noetic`.  
Maybe it also suitable for `melodic`.

venkisagunner93, 2020-05-14:
I imported the fix, ran it in our codebase. I'm not getting the error. Can I create a pull request to the melodic-devel branch referring to this fix you made @Barry-Xu-2018 ? I'm planning on testing thoroughly more and then create a pull request during the weekend anyway.

fujitatomoya, 2020-05-14:
PR is made for default branch, and if it is merged, then the fix is going to be back ported into each release branch.
BTW, we've confirmed that 3 days our production test pass without any error.

fujitatomoya, 2020-05-22:
@dirk-thomas 

i think that you can close this, https://github.com/ros/ros_comm/pull/1950 is merged.

we found this issue on our product, really appreciate for your quick response!!!

fujitatomoya, 2020-05-22:
I opened this! closing.

closed 2020-05-22 by fujitatomoya