ros/ros_comm #1955: Return type mismatch Subscriber.get_stats vs. Slave API getBusStats

pgorczak, 2020-05-19:
I ran into this while trying to use the [Slave API](http://wiki.ros.org/ROS/Slave_API) to query connection information via XMLRPC. The `getBusStats` call is supposed to return tuples of `[connectionId, bytesReceived, dropEstimate, connected]`. When querying a Python node, it returns 5 values instead. I think it passes on whatever [`Subscriber.get_stats`](https://github.com/ros/ros_comm/blob/noetic-devel/clients/rospy/src/rospy/topics.py#L672) returns, which is `[connectionId, bytesReceived, numSent, dropEstimate, connected]` according to the docstring.

dirk-thomas, 2020-05-19:
Please consider to update the wiki page to reflect the current behavior.

pgorczak, 2020-05-20:
Will do since the [cpp version](https://github.com/ros/ros_comm/blob/noetic-devel/clients/roscpp/src/libros/subscription.cpp#L110) seems to use the same format (although *connected* is always set to 0).

closed 2020-05-22 by pgorczak