ros/ros_comm #1958: XmlRpcValue ignores the value of `_doubleFormat`

peci1, 2020-05-20:
https://github.com/ros/ros_comm/search?q=_doubleformat&unscoped_q=_doubleformat
https://github.com/ros/ros_comm/search?q=getdoubleformat&unscoped_q=getdoubleformat

This value has no use at all.

    XmlRpc::XmlRpcValue::setDoubleFormat("%.2f");
    std::stringstream ss;
    ss << XmlRpc::XmlRpcValue(2.0);
    EXPECT_EQ("2.00", ss.str());

This fails. The output of `ss.str()` is "2".

dirk-thomas, 2020-07-13:
Please consider to contribute a pull request to fix the issue.

fujitatomoya, 2020-07-20:
@peci1 @dirk-thomas 

i do not think that this is bug in ros_comm but application. it just prints with "%.16g" format as default as expected. It needs to call method `setDoubleFormat("%.2f")` with object `XmlRpcValue(2.0)`.

dirk-thomas, 2020-07-20:
> It needs to call method `setDoubleFormat("%.2f")` with object `XmlRpcValue(2.0)`.

@fujitatomoya `setDoubleFormat` is `static` so it doesn't operate on a specific `XmlRpcValue` instance.

fujitatomoya, 2020-07-21:
> setDoubleFormat is static so it doesn't operate on a specific XmlRpcValue instance.

right, i was too early. my bad.

fujitatomoya, 2020-07-21:
@peci1 
CC: @dirk-thomas 

it would be nice if you could confirm https://github.com/ros/ros_comm/pull/2003 fixes the problem. (I did it locally but just in case.)

closed 2020-09-23 by dirk-thomas