ros/ros_comm #1960: ros::master::getTopic returns incomplete list of topics

facontidavide, 2020-05-21:
Hi,

I am using ros::master::getTopic and I noticed thta some topics are not present, when compared to those reported by **rostopic list**.

Most notably, the topics ending with `goal`  and `cancel` in an Actionlib server.

## How to reproduce

Lanch an action lib server:

      rosrun actionlib_tutorials averaging_server

Get the list of topics (in C++) with

      ros::master::getTopic(...)
 
My system is Melodic.

fujitatomoya, 2020-05-22:
> ros::master::getTopic(...)

correction, it is `ros::master::getTopics(...)`

confirmed the problem, here is how to reproduce.

1. setup melodic environment.
2. make catkin_workspace and clone the following tentative test branch.
```
git clone https://github.com/fujitatomoya/ros_tutorials/tree/test-20200522-getTopics-1960
```
3. catkin_make install
4. source setup.bash & roscore
5. issue following command and check output.
```
root@00f26df88c70:~/docker_ws/ros_catkin# rostopic list
/averaging/cancel
/averaging/feedback
/averaging/goal
/averaging/result
/averaging/status
/random_number
/rosout
/rosout_agg
root@00f26df88c70:~/docker_ws/ros_catkin# rosrun roscpp_tutorials talker 
[ INFO] [1590154759.166268011]: TopicInfo: name [/averaging/feedback], datatype[actionlib_tutorials/AveragingActionFeedback]
[ INFO] [1590154759.167518216]: TopicInfo: name [/averaging/status], datatype[actionlib_msgs/GoalStatusArray]
[ INFO] [1590154759.167569175]: TopicInfo: name [/rosout], datatype[rosgraph_msgs/Log]
[ INFO] [1590154759.167589365]: TopicInfo: name [/averaging/result], datatype[actionlib_tutorials/AveragingActionResult]
[ INFO] [1590154759.167651895]: TopicInfo: name [/rosout_agg], datatype[rosgraph_msgs/Log]
```

difference is
- action/cancel
- action/goal
- random_number

fujitatomoya, 2020-05-22:
with quick scan of the code, I do not think this is issue.

you can choose if publisher only or subscriber only or all via rostopic list [option],
```
root@00f26df88c70:~/docker_ws/ros_catkin# rostopic list -p
/averaging/feedback
/averaging/result
/averaging/status
/rosout
/rosout_agg
root@00f26df88c70:~/docker_ws/ros_catkin# rostopic list -s
/averaging/cancel
/averaging/goal
/random_number
/rosout
```

https://github.com/ros/ros_comm/blob/a2c4fbe597dd1dda0421bfa1dae5d4cd9c53a42b/clients/roscpp/include/ros/master.h#L101-L113
describes that `Get the list of topics that are being published by all nodes.`

facontidavide, 2020-05-22:
I need to access the entire list from C++, not the command line.

You are correct, the comment does specify that  `getTopics` is limited to the advertised topics only and will not provide the list of subscribed topics :(

I guess it is technically a "limitation", not an "issue" (still a little bit annoying for me, but that's life).

closed 2020-05-22 by facontidavide