ros/ros_comm #1966: [rosbag] Errors playing bags that contain latched and non-latched publishers of the same topic

mikepurvis, 2020-06-03:
We're experiencing an issue where we have multiple nodes publishing a topic, and when we go to play it back with rosbag, we get the error message that was introduced in #1544 when latched and unlatched publishers are created within the same process.

I think the fix should be in rosbag— to basically force all instances of a publisher to be latched if any of them are:

https://github.com/ros/ros_comm/blob/9162b32a42b5569ae42a94aa6426aafcb63021ae/tools/rosbag/src/player.cpp#L58-L73

dirk-thomas, 2020-06-03:
@meyerj FYI

mikepurvis, 2020-07-07:
@dirk-thomas @meyerj Do we have thoughts on a path forward here? There's potentially a similar problem here if multiple nodelets publish the same topic as latched/non-latched, though perhaps that is not as significant, since nodelets groups tend to be pipelines.

If we don't have a timeline for correcting this issue, I feel that #1544 should be reverted.

dirk-thomas, 2020-07-07:
> Do we have thoughts on a path forward here?

I don't have the bandwidth to work on a patch but I am happy to review any PR to address the regression.

> If we don't have a timeline for correcting this issue, I feel that #1544 should be reverted.

Reverting the patch as is would break API / ABI which isn't a good option. Instead the error message could be downgraded to a warning (if that is sufficient to make your use case work again)?

mikepurvis, 2020-07-07:
It's not the error that's the problem— it's the `return false`. :)

Looking again at the original change, I'm not really sure why the constraint against mixing latched and unlatched publishers was even introduced. It looks like most of the code should work fine for a scenario of mixed publisher types. I'll play with this a bit and see if I can add more tests that cover this scenario.

meyerj, 2020-07-07:
> It's not the error that's the problem— it's the `return false`. :)
>
> Looking again at the original change, I'm not really sure why the constraint against mixing latched and unlatched publishers was [...]

The problem is that external subscribers do not "see" individual publishers (or individual nodelets) if they connect to a node, but just the singleton publication. Before https://github.com/ros/ros_comm/pull/1544 it was possible to create a latched publisher and another volatile (non-latching) publisher, and the resulting publication was _latched_. Messages from both publishers, even from the second one, overwrote that latched sample and have been republished to new subscribers. On the other hand, if the volatile publisher was created first and then another latched one, the latch flag for the second publisher would have been ignored and the resulting publication was _not latching_. So in essence: the ROS protocol does not support mixing latching and non-latching publishers on the same topic. It it the topic that is latched or not, not the publisher. Because `rosbag play` runs as a single node, this restriction even applies to the whole deployment: If one node advertises a topic as latched, then no other node should advertise the topic as volatile.

With https://github.com/ros/ros_comm/pull/1544, the behavior is indeed a bit more defined and the restriction could be relaxed, but external subscribers like `rosbag record` still only see the publication and the latch flag of the first created publisher. So `rosbag play` can never reproduce the situation at the time the bag was recorded accurately as long as `rosbag record` cannot track which message has been publisher by which publisher instance (and not only by which node).

mikepurvis, 2020-07-08:
Thanks for the comment— let's carry on the discussion in #1991.

PR #1991 (2020-07-20): Allow mixing latched and unlatched publishers.
This is a naive change (so far) that just removes the check against mixing latched and unlatched publishers and updates the test added in #1544 so that it covers this scenario (it still passes). As far as I can tell, this should be totally safe and in line with the design intent of the original change:

1. Each Publisher instance now stores its own latched/unlatched status, and its own `last_message_` pointer, rather than this being at the TopicManager/singleton level.
2. When a new subscriber connects, the `Publication::peerConnect` method iterat [... cut]
closed 2020-07-08 by mikepurvis