ros/ros_comm #1972: ros::Publisher::publish doesn't accept basic types for std_msgs

brawner, 2020-06-12:
I am having trouble compiling a simple roscpp publisher that publishes raw doubles on a topic of std_msgs::Float64. In Melodic, you could publish the raw data value straight. Was this functionality intended to be changed?

Installed from Noetic binaries, with an upgraded workspace from previous distributions. The following example fails to compile for me. I could use help confirming this is not just in my head.

```
ros::Publisher publisher = nh.advertise<std_msgs::Float64>("numbers", 1);
...
double d = 1.0;
publisher.publish(d);
```

Whereas, the following works:
```
std_msgs::Float64 m;
m.data = 1.0;
publisher.publish(m);
```

Full example
```
#include <random>

#include <ros/ros.h>
#include <std_msgs/Float64.h>

int main(int argc, char **argv) {
  ros::init(argc, argv, "publisher");
  ros::NodeHandle nh;
  ros::Publisher publisher = nh.advertise<std_msgs::Float64>("numbers", 1);
  ros::Rate publish_rate(10);

  while (ros::ok()) {
    double d = 1.0;
    publisher.publish(d);
    publish_rate.sleep();
  }
}
```

```
In file included from /opt/ros/noetic/include/ros/serialization.h:37,
                 from /opt/ros/noetic/include/ros/publisher.h:34,
                 from /opt/ros/noetic/include/ros/node_handle.h:32,
                 from /opt/ros/noetic/include/ros/ros.h:45,
                 from /home/brawner/catkin_ws/src/myrobot/examples/src/example.cpp:3:
/opt/ros/noetic/include/ros/message_traits.h: In instantiation of ‘static const char* ros::message_traits::MD5Sum<M>::value(const M&) [with M = int]’:
/opt/ros/noetic/include/ros/message_traits.h:254:102:   required from ‘const char* ros::message_traits::md5sum(const M&) [with M = int]’
/opt/ros/noetic/include/ros/publisher.h:114:7:   required from ‘void ros::Publisher::publish(const M&) const [with M = int]’
/home/brawner/catkin_ws/src/myrobot/examples/src/example.cpp:21:24:   required from here
/opt/ros/noetic/include/ros/message_traits.h:125:14: error: request for member ‘__getMD5Sum’ in ‘m’, which is of non-class type ‘const int’
  125 |     return m.__getMD5Sum().c_str();
```

dirk-thomas, 2020-06-13:
> I am having trouble compiling a simple roscpp publisher that publishes raw doubles on a topic of std_msgs::Float64. In Melodic, you could publish the raw data value straight.

I don't think publishing the raw `data` value (instead of the message) is anything the API was ever intended to accept. These are the available `publish()` signatures in Melodic:
* https://github.com/ros/ros_comm/blob/ce22005d522628576908a68b1c747a12db492948/clients/roscpp/include/ros/publisher.h#L63-L64
* https://github.com/ros/ros_comm/blob/ce22005d522628576908a68b1c747a12db492948/clients/roscpp/include/ros/publisher.h#L95-L96

Afaik there have been no changes between Melodic and Noetic in this area.

brawner, 2020-06-13:
Interesting, I just tore apart my computer with bionic/melodic on it. I'll dig through it this weekend to see what was happening. But a package that compiled fine in Melodic broke and I was wondering if it was intentional.

Maybe the issue isn't with publish, but was there an implicit conversion from the raw type to the equivalent std_msg? I could've sworn that was  always a feature of std_msgs.

If there is no intent to support that sort of API in noetic, then maybe the point is sort of moot. I'll see what I can find after some investigation this weekend.

brawner, 2020-06-16:
Following up on this, it looks like the following compiles and works in Melodic:
```
double d = 1.0;
publisher.publish(d);
```

The following examples don't compile:

```
double d = 1.0;
std_msgs::Float64 m(d);
publisher.publish(m);
```

```
double d = 1.0;
std_msgs::Float64 m = d;
publisher.publish(m);
```

brawner, 2020-06-16:
So, not an implicit conversion, but an unintended feature of the publish API looks like.

dirk-thomas, 2020-06-16:
> ```
> double d = 1.0;
> publisher.publish(d);
> ```

The `publish()` method is templated on the argument. Probably this case is lucky that the message type and the `double` type are a perfect match of the memory layout.

I am not aware of any change in that area of the code. If you could point to a change which caused this I am more than happy to consider a patch to restore the previous behavior.

To narrow it down I would suggest to try using the newer compiler (from Ubuntu Focal) to build the example with the Melodic code. Maybe that case already fails if the change is due a change in compiler strictness...

brawner, 2020-06-16:
I think the issue has come down to a matter of Debug build vs RelWithDebInfo build. The ROS_ASSERT_MSG is not compiled in during debug mode. However, serialization has been defined for simple types here:
https://github.com/ros/roscpp_core/blob/eabafec8f9b554d5389a1aa1b3c8145c07faaefb/roscpp_serialization/include/ros/serialization.h#L203

So it can compile in release mode just fine, but not debug. For some reason, my catkin workspace was building in debug on noetic.

Looks like the behavior hasn't changed that I can tell. Closing.

closed 2020-06-16 by brawner

tahsinkose, 2020-11-25:
@brawner, actually assertions are being compiled in debug mode. Release modes of CMake specify `-DNDEBUG` implicitly, hence you don't get any errors in release modes.