ros/ros_comm #1975: New Kinetic Release

DLu, 2020-06-20:
There are a handful of backports on the kinetic branch [dating back to early 2019](https://github.com/ros/ros_comm/commit/4c4ecec73e647512fd086ab880f93b38b71330c2) that have not been released for kinetic. 

It would be useful to cut a new release for these features.

DLu, 2020-07-06:
Paging @dirk-thomas

dirk-thomas, 2020-07-08:
I understand the desire to get a new Kinetic release of `ros_comm` out but in the light of #1496 and me being the only person at Open Robotics doing ROS 1 maintenance work on ROS core packages I unfortunately can't give an estimate when I will be able to catch up with Kinetic backports and a release. The little amount of time I can use towards maintenance is usually used by work for newer distros. I know this is not a satisfying answer but I just wanted to describe how it is and give a realistic expectation.

DLu, 2020-07-09:
@dirk-thomas I think we both understand the burdens of maintaining core ROS software. However, I'm not asking for additional backports, but just release of the repo as-is. I do not have the permissions to make the release, otherwise I would do it myself. I believe not doing the release would be a disservice to past-Dirk who [already](https://github.com/ros/ros_comm/pull/1685) [worked](https://github.com/ros/ros_comm/pull/1789) [hard](https://github.com/ros/ros_comm/pull/1730) [reviewing and](https://github.com/ros/ros_comm/pull/1576) [merging previous PRs](https://github.com/ros/ros_comm/pull/1578)

DLu, 2020-07-12:
It appears as though I *have* been given permissions to make the release. Any objection to me doing that next week?

dirk-thomas, 2020-07-14:
> It appears as though I have been given permissions to make the release. Any objection to me doing that next week?

I don't know if anything has changed regarding permissions recently. Please do no make a release yourself without the maintainer of the repository. I will do a release when I find the time and will try to get to it within a week.

DLu, 2020-07-29:
[image]

dirk-thomas, 2020-08-04:
See #2015 for a list of all considered and backported changes.

Waiting if #1284 gets an update to be included in the upcoming patch release.

dirk-thomas, 2020-08-10:
And finally the new patch release is on the way: ros/rosdistro#26148.

closed 2020-08-10 by dirk-thomas

DLu, 2020-08-11:
[image]