ros/ros_comm #1979: Command Line "rosparam set " can not work

Joy-Zhang-0303, 2020-06-26:
Hello, 

I need someone's help about "rosparam set".

When I use "rosparam set" to change one parameter, 
- I can confirm the changed paramater through "rosparam get".
- however, actually, the parameter was **not changed.** 

you can see the parameter results from command line and dynamic-reconfigure GUI.
[image]

Joy-Zhang-0303, 2020-06-26:
What I want to do is: after I revise the parameter by the commnd line, the chaged parameter coule be reflectted(used) soon.
As the breaf background, I need to change local planner name by command line when I run the robot simulation. As the test,  after I confirm the "rosparam set" by using other parameters, I found it doesn't work.

fujitatomoya, 2020-06-26:
1st of all, ros parameter is server-client architecture, you could refer to http://wiki.ros.org/Parameter%20Server.

> I can confirm the changed paramater through "rosparam get".

because it accesses to master via xmlrpc to read the parameter.

> however, actually, the parameter was not changed.

parameters are stored in the rosmaster, not distributed to every single node. so you need to read it from the rosmaster.

in your use case, why do you not use topic instead of parameter?

fujitatomoya, 2020-06-26:
@Joy-Zhang-0303 

github/issues are to register issues and problems, i think [ROS Answers](https://answers.ros.org/questions/) would be the best place to ask this kind of information.

knorth55, 2020-06-26:
@Joy-Zhang-0303 
if you want to get `dynamic_reconfigure` parametes, you should use `dynamic_reconfigure dynparam get`, not `rosparam get`.

```bash
rosrun dynamic_reconfigure dynparam get /move_base/local_costmap
```

Joy-Zhang-0303, 2020-06-26:
@fujitatomoya 

Thanks for your comment firstly.

For your comment as above, I will reply respectivately.

1. parameters are stored in the rosmaster, not distributed to every si1ngle node. so you need to read it from the rosmaster.
A: I am sorry that I didn't understand what do you mean "so you need to read it from the rosmaster."  
In other words, could you please explain how could we let the ROS node read the parameters stored in parameter server ?

2. in your use case, why do you not use topic instead of parameter?
A: My purpoes is to change Local Planner Type when running simulation. In detail, there are 2 planners(A and B), in .yaml file, the default one is A. To research the performance of 2 planners, I need to change planner A to B by manual. 
So, that is why I want to use rosparam to change it. 
Of cource, the dynamic_reconfigure is another choice to select. But currently, I am trying to find one simple way.

Joy-Zhang-0303, 2020-06-26:
@knorth55 

Thanks for you. 
The parameter that I want to adjust now is not available in dynamic_reconfigure. 
As I replied for fujitatomoya-san, of cource, we can write the .cfg file to make that parameter available in the dynamic_reconfigure GUI, but currently, I am trying to find one simpler way.

fujitatomoya, 2020-06-26:
@Joy-Zhang-0303 

> I am sorry that I didn't understand what do you mean "so you need to read it from the rosmaster."
In other words, could you please explain how could we let the ROS node read the parameters stored in parameter server ?

i mean call the API as sample shows to read the parameter, [Getting_Parameters](http://wiki.ros.org/roscpp/Overview/Parameter%20Server#Getting_Parameters)

> My purpoes is to change Local Planner Type when running simulation. In detail, there are 2 planners(A and B), in .yaml file, the default one is A. To research the performance of 2 planners, I need to change planner A to B by manual.
So, that is why I want to use rosparam to change it.

okay i was too early, yet i am not sure about your requirement. how about the following?

1. set the parameter.
2. get the parameter and then start Local Planner based on the parameter.

closed 2020-06-26 by dirk-thomas