ros/ros_comm #1987: Node process die when running in arm64v8 (qemu) docker

phamelin, 2020-06-29:
**Configuration**

- ROS Melodic
- Ubuntu 18.04

**Issue details**

I'm unable to load a node when running ROS inside a qemu arm64v8 docker. The node process dies shortly after being started. The host computer is running Ubuntu 18.04 64 bits. The problem can be easily reproduced as follow:

```
# Setup qemu to run arm docker on x86 host
sudo apt-get install qemu binfmt-support qemu-user-static
docker run --rm --privileged multiarch/qemu-user-static --reset -p yes

# Launch arm64v8 container with ros image
docker run -it arm64v8/ros:melodic-ros-base /bin/bash

# Install and run roscpp tutorial
apt-get update && apt-get install ros-melodic-roscpp-tutorials
source /opt/ros/melodic/setup.bash
roslaunch roscpp_tutorials talker_listener.launch

```
Here is the output:

```
... logging to /root/.ros/log/a72e1788-b002-11ea-9127-3f45a6660ee1/roslaunch-8dd91dac75d2-2141.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://8dd91dac75d2:37539/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    listener (roscpp_tutorials/listener)
    talker (roscpp_tutorials/talker)

auto-starting new master
process[master]: started with pid [2166]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a72e1788-b002-11ea-9127-3f45a6660ee1
process[rosout-1]: started with pid [2183]
started core service [/rosout]
process[listener-2]: started with pid [2188]
[rosout-1] process has died [pid 2183, exit code 127, cmd /opt/ros/melodic/lib/rosout/rosout __name:=rosout __log:=/root/.ros/log/a72e1788-b002-11ea-9127-3f45a6660ee1/rosout-1.log].
log file: /root/.ros/log/a72e1788-b002-11ea-9127-3f45a6660ee1/rosout-1*.log
[rosout-1] restarting process
process[talker-3]: started with pid [2195]
[ INFO] [1592333610.205317895]: hello world 0
qemu:handle_cpu_signal received signal outside vCPU context @ pc=0x6005fc3a
qemu:handle_cpu_signal received signal outside vCPU context @ pc=0x60001737
[ INFO] [1592333610.301632331]: hello world 1
process[rosout-1]: started with pid [2202]
[ INFO] [1592333610.400877386]: hello world 2
[listener-2] process has died [pid 2188, exit code 127, cmd /opt/ros/melodic/lib/roscpp_tutorials/listener __name:=listener __log:=/root/.ros/log/a72e1788-b002-11ea-9127-3f45a6660ee1/listener-2.log].
log file: /root/.ros/log/a72e1788-b002-11ea-9127-3f45a6660ee1/listener-2*.log
[ INFO] [1592333610.500888051]: hello world 3
[ INFO] [1592333610.600877653]: hello world 4
qemu:handle_cpu_signal received signal outside vCPU context @ pc=0x6005fc3a
qemu:handle_cpu_signal received signal outside vCPU context @ pc=0x60001737
[talker-3] process has died [pid 2195, exit code 127, cmd /opt/ros/melodic/lib/roscpp_tutorials/talker __name:=talker __log:=/root/.ros/log/a72e1788-b002-11ea-9127-3f45a6660ee1/talker-3.log].
log file: /root/.ros/log/a72e1788-b002-11ea-9127-3f45a6660ee1/talker-3*.log

```
The crash seems to be related to these messages:
`qemu:handle_cpu_signal received signal outside vCPU context @ pc=0x6005fc3a`
and I didn't find any relevant solution on Google.

escobarhector, 2021-11-04:
I got the same issue!

jonathankehler, 2022-05-21:
Ditto

chacalnoir, 2023-08-21:
I see a similar issue on Ubuntu 20.04, ROS Noetic, arm32v7 (qemu) docker. I don't see the `qemu:handle_cpu_signal received signal outside vCPU context @ pc=0x6005fc3a` line, but get the same crashing. See below for an abbrieviated output.
`[ INFO] [1692630267.084173658]: hello world 0
[listener-2] process has died [pid 3027, exit code -11, cmd /opt/ros/noetic/lib/roscpp_tutorials/listener __name:=listener __log:=/home/.ros/log/0407e92e-4034-11ee-9ae1-0242ac110002/listener-2.log].
log file: /home/.ros/log/0407e92e-4034-11ee-9ae1-0242ac110002/listener-2*.log
[ INFO] [1692630267.183513624]: hello world 1
[rosout-1] process has died [pid 3022, exit code -11, cmd /opt/ros/noetic/lib/rosout/rosout __name:=rosout __log:=/home/.ros/log/0407e92e-4034-11ee-9ae1-0242ac110002/rosout-1.log].
log file: /home/.ros/log/0407e92e-4034-11ee-9ae1-0242ac110002/rosout-1*.log
[rosout-1] restarting process
process[rosout-1]: started with pid [3047]
[ INFO] [1692630267.282332414]: hello world 2
[talker-3] process has died [pid 3030, exit code -11, cmd /opt/ros/noetic/lib/roscpp_tutorials/talker __name:=talker __log:=/home/.ros/log/0407e92e-4034-11ee-9ae1-0242ac110002/talker-3.log].
log file: /home/.ros/log/0407e92e-4034-11ee-9ae1-0242ac110002/talker-3*.log
`

jspricke, 2023-08-22:
This sounds like a bug in (an old version of) Docker/qemu-user-static/Ubuntu 20.04/.. and unrelated to this project. At least I can't reproduce it in an up to date (Debian unstable) DebVM:
```
debvm-create -a arm64 -- --include=git,ca-certificates,catkin,make,g++,ros-message-generation,libroscpp-dev,python3-roslaunch
debvm-run
# inside
git clone https://github.com/ros/ros_tutorials
cd ros_tutorials/roscpp_tutorials/
mkdir build
cd build
cmake ..
make
. devel/setup.sh
roslaunch roscpp_tutorials talker_listener.launch
```

chacalnoir, 2023-08-22:
Okay, that answers my question. Looks like the issue has existed for a while, but is primarily with the underlying system (Docker + qemu), which is fixed with newer versions. Since I can't run the newest version of Ubuntu currently (so don't have access to DebVM), I can't replicate a working environment for the work I am doing, but that is irrelevant here.

Thank you for taking the time to test and answer, as that confirms my suspicion of where the issue likely exists, as well as a solution to it once the updated environment is available to me for this task.

KanonWY, 2024-01-31:
Okay, any solutions for this question?

amefabris, 2024-04-05:
I am facing the same issue on a Ubuntu 22.04 (x86_64) machine with a docker container arm64v8 and ROS Noetic. The container works OK, but the node process dies shortly after being started, even `rosbag play` dies straight away with no errors while on the master side I get 
```
[rosout-1] process has died [pid 206, exit code -11, cmd /opt/ros/noetic/lib/rosout/rosout __name:=rosout __log:=/root/.ros/log/237785f0-f31a-11ee-a184-6c1ff7002326/rosout-1.log].
log file: /root/.ros/log/237785f0-f31a-11ee-a184-6c1ff7002326/rosout-1*.log
```