ros/ros_comm #1989: boost::recursive_mutex assert fails in ros::Connection::drop()  when Connection drops

meng-atomic, 2020-07-03:
# Core Stack

#0 0x00007f2d7c657428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1 0x00007f2d7c65902a in __GI_abort () at abort.c:89
#2 0x00007f2d7c64fbd7 in __assert_fail_base (fmt=<optimized out>, assertion=assertion@entry=0x7c8095 "!pthread_mutex_lock(&m)",
    file=file@entry=0x7c60d0 "/usr/include/boost/thread/pthread/recursive_mutex.hpp", line=line@entry=113,
    function=function@entry=0x7cbdc0 <boost::recursive_mutex::lock()::__PRETTY_FUNCTION__> "void boost::recursive_mutex::lock()")
    at assert.c:92
#3 0x00007f2d7c64fc82 in __GI___assert_fail (assertion=assertion@entry=0x7c8095 "!pthread_mutex_lock(&m)",
    file=file@entry=0x7c60d0 "/usr/include/boost/thread/pthread/recursive_mutex.hpp", line=line@entry=113,
    function=function@entry=0x7cbdc0 <boost::recursive_mutex::lock()::__PRETTY_FUNCTION__> "void boost::recursive_mutex::lock()")
    at assert.c:101
#4 0x00000000005e1b7e in boost::recursive_mutex::lock (this=<optimized out>)
    at /usr/include/boost/thread/pthread/recursive_mutex.hpp:113
#5 boost::unique_lock<boost::recursive_mutex>::lock (this=0x7f2d74663270) at /usr/include/boost/thread/lock_types.hpp:346
#6 0x00007f2d80e93085 in ros::Connection::drop(ros::Connection::DropReason) () from /opt/ros/kinetic/lib/libroscpp.so
#7 0x00007f2d80f0e9b5 in ros::TransportPublisherLink::drop() () from /opt/ros/kinetic/lib/libroscpp.so
#8 0x00007f2d80f365f1 in ros::Subscription::pubUpdate(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) ()
   from /opt/ros/kinetic/lib/libroscpp.so
#9 0x00007f2d80ec0193 in ros::TopicManager::pubUpdate(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) () from /opt/ros/kinetic/lib/libroscpp.so
#10 0x00007f2d80ec0676 in ros::TopicManager::pubUpdateCallback(XmlRpc::XmlRpcValue&, XmlRpc::XmlRpcValue&) ()
   from /opt/ros/kinetic/lib/libroscpp.so

# Crash Site
https://github.com/ros/ros_comm/blob/c7f7598e8a49b1b4eb6f3e42f698147e151a811d/clients/roscpp/src/libros/connection.cpp#L330

# Analysis
## boost::recursive_mutex assert_fail
https://github.com/ros/ros_comm/blob/c7f7598e8a49b1b4eb6f3e42f698147e151a811d/clients/roscpp/src/libros/connection.cpp#L330
When the scoped_lock tries to lock the mutex, the assertion happened. Why? Something must be wrong with the mutex, namely **drop_mutex_**. Since the drop_mutex_ is a member of class Connection, it makes sense to infer that the drop_mutex_ has already been destructed with the destruction of an object of class Connection.    
When the code is running, the following line will be executed:
`boost::recursive_mutex::scoped_lock lock(this->drop_mutex_);`
The pointer points to a memory address, an Object of Connection should be stored there.

fujitatomoya, 2020-07-05:
@meng-atomic 

thanks for the insights!

one question which version of ROS did you meet this core stack? like noetic or melodic?
and can you reproduce this issue with [SSCCE](http://sscce.org/)?

this could be related to the followings, these have been addressed with Noetic release.

- https://github.com/ros/ros_comm/issues/1949
- https://github.com/ros/ros_comm/issues/1939

and in your theory, the problem could happen at any stack with using Connection object, right?

meng-atomic, 2020-07-06:
@fujitatomoya I've reproduced the issue to make it short and clear. #1949 #1939 and this issue are all caused by the drop of a connection, but crased in different sites for different reasons.
TransportPublisherLink::connection_ is  not thread safe, so, yes , the problem could happen with using Connection object.

fujitatomoya, 2020-07-06:
@meng-atomic 

> I've reproduced the issue to make it short and clear.

is that possible if you could share that test sample with us?