ros/ros_comm #1994: cannot import custom message in python

tzf230201, 2020-07-11:
this is the node.
i use my own custom message

```
#!/usr/bin/env python
import rospy
import cv2
from std_msgs.msg import String
from sensor_msgs.msg import Image
from diff_drive_lf import line_sensor
from cv_bridge import CvBridge, CvBridgeError
import sys
import numpy as np

bridge = CvBridge()

def image_callback(ros_image):
  print 'got an image'
  global bridge
  #convert ros_image into an opencv-compatible image
  try:
    cv_image = bridge.imgmsg_to_cv2(ros_image, "bgr8")
  except CvBridgeError as e:
      print(e)

  #from now on, you can work exactly like with opencv
  cv_image = cv2.resize(cv_image, (0,0), fx=0.5,fy=0.5)
  cv_image = cv2.cvtColor(cv_image, cv2.COLOR_BGR2GRAY)

  sens = line_sensor()
  sens.lineSensor1 = cv_image[399][25]
  sens.lineSensor2 = cv_image[399][75]
  sens.lineSensor3 = cv_image[399][125]
  sens.lineSensor4 = cv_image[399][175]
  sens.lineSensor5 = cv_image[399][225]
  sens.lineSensor6 = cv_image[399][275]
  sens.lineSensor7 = cv_image[399][325]
  sens.lineSensor8 = cv_image[399][375]

  pub = rospy.Publisher('/lineSensor', line_sensor, queue_size=20)
  pub.publish(sens)

  cv2.imshow("lf_camera", cv_image)

  cv2.waitKey(3)

  
def main(args):
  rospy.init_node('image_converter', anonymous=True)
  image_sub = rospy.Subscriber("/lf_camera/image_raw",Image, image_callback)

  try:
    rospy.spin()
  except KeyboardInterrupt:
    print("Shutting down")
  cv2.destroyAllWindows()

if __name__ == '__main__':
    main(sys.argv)
```
but when i rosrun diff_drive_lf line_tracker.py , i got this..

```
Traceback (most recent call last):
  File "line_tracker.py", line 6, in <module>
    from diff_drive_lf import line_sensor
ImportError: No module named diff_drive_lf
```
but the message appears when i call rosmsg list
`diff_drive_lf/line_sensor`

my Cmakelist is:
```
cmake_minimum_required(VERSION 3.0.2)
project(diff_drive_lf)

find_package(catkin REQUIRED COMPONENTS
  joint_state_publisher
  joint_state_publisher_gui
  robot_state_publisher
  rviz
  xacro
  controller_manager
  gazebo_ros
  gazebo_ros_control
  joint_state_controller
  rqt_robot_steering
  diff_drive_controller
  OpenCV
  roscpp
  rospy
  geometry_msgs
  std_msgs
  message_generation
)

catkin_package(
  CATKIN_DEPENDS message_runtime
)

add_message_files(
   FILES
   line_sensor.msg 
)

include_directories(
  ${catkin_INCLUDE_DIRS}
  ${OpenCV_INCLUDE_DIRS}
)

add_executable(controller config/controller.cpp)
target_link_libraries(controller ${catkin_LIBRARIES})

```
the the package.xml is
```

<?xml version="1.0"?>
<package format="2">
  <name>diff_drive_lf</name>
  <version>0.0.0</version>
  <description>The diff_drive_lf package</description>

  <!-- One maintainer tag required, multiple allowed, one person per tag -->
  <!-- Example:  -->
  <!-- <maintainer email="jane.doe@example.com">Jane Doe</maintainer> -->
  <maintainer email="apon@todo.todo">apon</maintainer>

  <!-- One license tag required, multiple allowed, one license per tag -->
  <!-- Commonly used license strings: -->
  <!--   BSD, MIT, Boost Software License, GPLv2, GPLv3, LGPLv2.1, LGPLv3 -->
  <license>TODO</license>

  <!-- Url tags are optional, but multiple are allowed, one per tag -->
  <!-- Optional attribute type can be: website, bugtracker, or repository -->
  <!-- Example: -->
  <!-- <url type="website">http://wiki.ros.org/diff_drive_lf</url> -->

  <!-- Author tags are optional, multiple are allowed, one per tag -->
  <!-- Authors do not have to be maintainers, but could be -->
  <!-- Example: -->
  <!-- <author email="jane.doe@example.com">Jane Doe</author> -->

  <!-- The *depend tags are used to specify dependencies -->
  <!-- Dependencies can be catkin packages or system dependencies -->
  <!-- Examples: -->
  <!-- Use depend as a shortcut for packages that are both build and exec dependencies -->
  <!--   <depend>roscpp</depend> -->
  <!--   Note that this is equivalent to the following: -->
  <!--   <build_depend>roscpp</build_depend> -->
  <!--   <exec_depend>roscpp</exec_depend> -->
  <!-- Use build_depend for packages you need at compile time: -->
  <!--   <build_depend>message_generation</build_depend> -->
  <!-- Use build_export_depend for packages you need in order to build against this package: -->
  <!--   <build_export_depend>message_generation</build_export_depend> -->
  <!-- Use buildtool_depend for build tool packages: -->
  <!--   <buildtool_depend>catkin</buildtool_depend> -->
  <!-- Use exec_depend for packages you need at runtime: -->
  <!--   <exec_depend>message_runtime</exec_depend> -->
  <!-- Use test_depend for packages you need only for testing: -->
  <!--   <test_depend>gtest</test_depend> -->
  <!-- Use doc_depend for packages you need only for building documentation: -->
  <!--   <doc_depend>doxygen</doc_depend> -->
  <buildtool_depend>catkin</buildtool_depend>
  <build_depend>joint_state_publisher</build_depend>
  <build_depend>joint_state_publisher_gui</build_depend>
  <build_depend>robot_state_publisher</build_depend>
  <build_depend>rviz</build_depend>
  <build_depend>xacro</build_depend>
  <build_depend>controller_manager</build_depend>
  <build_depend>gazebo_ros</build_depend>
  <build_depend>gazebo_ros_control</build_depend>
  <build_depend>joint_state_controller</build_depend>
  <build_depend>rqt_robot_steering</build_depend>
  <build_depend>diff_drive_controller</build_depend>
  <build_depend>roscpp</build_depend>
  <build_depend>rospy</build_depend>
  <build_depend>std_msgs</build_depend>
  <build_depend>geometry_msgs</build_depend>
  <build_depend>message_generation</build_depend>

  <build_export_depend>joint_state_publisher</build_export_depend>
  <build_export_depend>joint_state_publisher_gui</build_export_depend>
  <build_export_depend>robot_state_publisher</build_export_depend>
  <build_export_depend>rviz</build_export_depend>
  <build_export_depend>xacro</build_export_depend>
  <build_export_depend>controller_manager</build_export_depend>
  <build_export_depend>gazebo_ros</build_export_depend>
  <build_export_depend>gazebo_ros_control</build_export_depend>
  <build_export_depend>joint_state_controller</build_export_depend>
  <build_export_depend>rqt_robot_steering</build_export_depend>
  <build_export_depend>diff_drive_controller</build_export_depend>
  <build_export_depend>roscpp</build_export_depend>
  <build_export_depend>rospy</build_export_depend>
  <build_export_depend>std_msgs</build_export_depend>
  <build_export_depend>geometry_msgs</build_export_depend>

  <exec_depend>joint_state_publisher</exec_depend>
  <exec_depend>joint_state_publisher_gui</exec_depend>
  <exec_depend>robot_state_publisher</exec_depend>
  <exec_depend>rviz</exec_depend>
  <exec_depend>xacro</exec_depend>
  <exec_depend>controller_manager</exec_depend>
  <exec_depend>gazebo_ros</exec_depend>
  <exec_depend>gazebo_ros_control</exec_depend>
  <exec_depend>joint_state_controller</exec_depend>
  <exec_depend>rqt_robot_steering</exec_depend>
  <exec_depend>diff_drive_controller</exec_depend>
  <exec_depend>roscpp</exec_depend>
  <exec_depend>rospy</exec_depend>
  <exec_depend>std_msgs</exec_depend>
  <exec_depend>geometry_msgs</exec_depend>
  <exec_depend>message_runtime</exec_depend>

  
  <!-- The export tag contains other, unspecified, tags -->
  <export>
    <!-- Other tools can request additional information be placed here -->

  </export>
</package>

```
I have seen some references from the internet but have not found a solution...

could anyone help me? please

dirk-thomas, 2020-07-13:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2020-07-13 by dirk-thomas

tzf230201, 2020-07-15:
i am sorry for my mistake