ros/ros_comm #1995: building roscpp error "‘_2’ was not declared in this scope"

HemaZ, 2020-07-13:
hello, 
I've been trying to build roscpp on Raspian buster and RPI4 but it fails due to the following error.

```
In file included from /home/pi/catkin_ws/src/ros_comm/clients/roscpp/src/libros/wall_timer.cpp:29:
/home/pi/catkin_ws/src/ros_comm/clients/roscpp/include/ros/timer_manager.h:419:72: error: ‘_2’ was not declared in this scope
     waiting_.sort(boost::bind(&TimerManager::waitingCompare, this, _1, _2));

```
I've BOOST_LIB_VERSION "1_67" installed 
Thanks

mikepurvis, 2020-07-13:
Buster _should_ work per https://www.ros.org/reps/rep-0003.html#noetic-ninjemys-may-2020-may-2025, but out of curiosity, does it work if you set up your workspace with the Melodic branches?

HemaZ, 2020-07-13:
> Buster _should_ work per https://www.ros.org/reps/rep-0003.html#noetic-ninjemys-may-2020-may-2025, but out of curiosity, does it work if you set up your workspace with the Melodic branches?

I was able to install it by using melodic repos. but some Python3 packages I had to install manually 
I've documented the whole installation process here 

https://gist.github.com/HemaZ/fb1a144a20ace1fbf9429aaedaf919f0

dirk-thomas, 2020-07-13:
> I was able to install it by using melodic repos. but some Python3 packages I had to install manually

dirk-thomas, 2020-07-13:
> I was able to install it by using melodic repos. but some Python3 packages I had to install manually

Only Noetic officially supports Python 3. Melodic targets Python 2.7.

Also installing the message generation packages via `pip` is not recommended / supported.

HemaZ, 2020-07-13:
I think this error was due to having multiple Boost Library on my Raspbian. But it now works 

> > I was able to install it by using melodic repos. but some Python3 packages I had to install manually
> 
> Only Noetic officially supports Python 3. Melodic targets Python 2.7.
> 
> Also installing the message generation packages via `pip` is not recommended / supported.

The installation was complaining about not finding the message generation packages but installing them manually solved the issues for me.

HemaZ, 2020-07-13:
about the main issues stated above, I think it was due to multiple versions of Boost installed on the system. as I had boost 1_67 and boost 1_73 (installed from source) . installing on a fresh raspbian image worked for me. I think this issue could be closed.

dirk-thomas, 2020-07-13:
> I think this issue could be closed.

Since you opened the ticket you should be able to close it yourself :+1:

closed 2020-07-13 by HemaZ