ros/ros_comm #1997: New Melodic release

smipi1, 2020-07-14:
Would it be possible for a new Melodic release?
The Ubuntu bionic package is stuck at 1.4.1 and fixes for a few nasty bugs only landed in 1.4.4.

dirk-thomas, 2020-07-21:
> The Ubuntu bionic package is stuck at 1.4.1 and fixes for a few nasty bugs only landed in 1.4.4.

The latest released version in Melodic is 1.14.6 (from May 29th): https://github.com/ros/rosdistro/blob/4d65c3c8e7cb173a5f7a0fa584e36fb6470c134a/melodic/distribution.yaml#L8927 / ros/rosdistro#25202.

dirk-thomas, 2020-07-24:
There are no significant patches in the upcoming Melodic patch release 1.14.7: see #2008.

dirk-thomas, 2020-07-28:
Actually #1932 was also merged for the upcoming Melodic patch release which contains a significant patch.

smipi1, 2020-07-29:
Thanks for the update!

dirk-thomas, 2020-07-31:
#1684 and #2014 were also merged for the upcoming Melodic patch release - with a significant fix.

dirk-thomas, 2020-07-31:
New Melodic release 1.14.7 in flight: ros/rosdistro#26012.

closed 2020-07-31 by dirk-thomas