ros/ros_comm #1998: ropy: Subscriper shows "error: can't start new thread"

mistoll, 2020-07-15:
There are multiple publishers of the /cmd_vel topic, which may start and stop frequently. Typical run time is 1-20 minutes.
Two nodes are listening to /cmd_vel. On is using roscpp, one is using rospy. The roscpp node is working great.

After working correctly for some days, the rospy node starts to missbehave. It shows "error: can't start new thread" errors in the log.

```
[rospy.internal][INFO] 2020-07-09 12:14:06,826: topic[/rosout] adding connection to [/connect_log], count 1
[rospy.internal][INFO] 2020-07-09 12:15:14,213: topic[/base/cmd_vel] adding connection to [http://chloe:44039/], count 0
[rospy.internal][INFO] 2020-07-09 12:15:33,540: topic[/base/cmd_vel] adding connection to [http://chloe:34397/], count 1
[rospy.internal][WARNING] 2020-07-09 12:18:13,355: Unknown error initiating TCP/IP socket to chloe:36581 (http://chloe:44039/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 562, in connect
    self.read_header()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 657, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosgraph/network.py", line 357, in read_ros_handshake_header
    d = sock.recv(buff_size)
error: [Errno 104] Connection reset by peer

[rospy.internal][INFO] 2020-07-09 12:18:13,355: topic[/base/cmd_vel] removing connection to http://chloe:44039/
[rospy.internal][ERROR] 2020-07-09 12:18:14,298: Unable to initiate TCP/IP socket to chloe:41451 (http://chloe:34397/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 562, in connect
    self.read_header()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 657, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 600, in _validate_header
    raise TransportInitError("remote error reported: %s"%header['error'])
TransportInitError: remote error reported: node shutting down

[rospy.internal][WARNING] 2020-07-09 12:18:15,301: Unknown error initiating TCP/IP socket to chloe:41451 (http://chloe:34397/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 560, in connect
    self.socket.connect((dest_addr, dest_port))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 111] Connection refused

[rospy.internal][INFO] 2020-07-09 12:18:15,301: topic[/base/cmd_vel] removing connection to http://chloe:34397/
[rospy.internal][INFO] 2020-07-09 12:18:17,330: topic[/base/cmd_vel] adding connection to [http://chloe:43949/], count 0
[rospy.internal][INFO] 2020-07-09 12:20:45,532: topic[/base/cmd_vel] adding connection to [http://chloe:36297/], count 1
[rospy.internal][ERROR] 2020-07-09 12:22:09,482: Unable to initiate TCP/IP socket to chloe:37193 (http://chloe:36297/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 562, in connect
    self.read_header()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 657, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 600, in _validate_header
    raise TransportInitError("remote error reported: %s"%header['error'])
TransportInitError: remote error reported: node shutting down

[rospy.internal][WARNING] 2020-07-09 12:22:10,484: Unknown error initiating TCP/IP socket to chloe:37193 (http://chloe:36297/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 560, in connect
    self.socket.connect((dest_addr, dest_port))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 111] Connection refused

...

[rospy.internal][WARNING] 2020-07-13 17:28:34,314: Unknown error initiating TCP/IP socket to chloe:45711 (http://chloe:35447/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 560, in connect
    self.socket.connect((dest_addr, dest_port))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 111] Connection refused

[rospy.internal][INFO] 2020-07-13 17:28:34,315: topic[/base/cmd_vel] removing connection to http://chloe:35447/
[rospy.internal][INFO] 2020-07-13 17:28:58,336: topic[/base/cmd_vel] adding connection to [http://chloe:46403/], count 0
[rospy.internal][INFO] 2020-07-13 17:34:24,161: topic[/base/cmd_vel] adding connection to [http://chloe:41923/], count 1
[rospy.internal][WARNING] 2020-07-13 17:56:59,924: Unknown error initiating TCP/IP socket to chloe:42671 (http://chloe:41923/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 561, in connect
    self.write_header()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 642, in write_header
    self.stat_bytes += write_ros_handshake_header(sock, protocol.get_header_fields())
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosgraph/network.py", line 417, in write_ros_handshake_header
    sock.sendall(s)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 104] Connection reset by peer

[rospy.internal][INFO] 2020-07-13 17:56:59,925: topic[/base/cmd_vel] removing connection to http://chloe:41923/
[rospy.internal][WARNING] 2020-07-13 17:59:13,475: Unknown error initiating TCP/IP socket to chloe:39071 (http://chloe:46403/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 562, in connect
    self.read_header()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 657, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosgraph/network.py", line 357, in read_ros_handshake_header
    d = sock.recv(buff_size)
error: [Errno 104] Connection reset by peer

[rospy.internal][INFO] 2020-07-13 17:59:13,475: topic[/base/cmd_vel] removing connection to http://chloe:46403/
[rospy.internal][INFO] 2020-07-13 17:59:19,482: topic[/base/cmd_vel] adding connection to [http://chloe:37075/], count 0
[rosout][WARNING] 2020-07-14 04:41:09,766: Inbound TCP/IP connection failed: can't start new thread
[rospy.internal][ERROR] 2020-07-14 04:41:09,779: Inbound TCP/IP connection failed:
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 332, in _tcp_server_callback
    err_msg = self.service_connection_handler(sock, client_addr, header)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 256, in service_connection_handler
    t.start()
  File "/usr/lib/python2.7/threading.py", line 736, in start
    _start_new_thread(self.__bootstrap, ())
error: can't start new thread

[rosout][WARNING] 2020-07-14 05:29:23,936: Inbound TCP/IP connection failed: can't start new thread
[rospy.internal][ERROR] 2020-07-14 05:29:23,937: Inbound TCP/IP connection failed:
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 332, in _tcp_server_callback
    err_msg = self.service_connection_handler(sock, client_addr, header)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 256, in service_connection_handler
    t.start()
  File "/usr/lib/python2.7/threading.py", line 736, in start
    _start_new_thread(self.__bootstrap, ())
error: can't start new thread
...
```

Could this be a bug in rospy?

peci1, 2020-10-01:
It is possible you are blocked by a limit on the number of threads. We've faced this now on a system with a few hundreds of nodes.

What does this show to you?

    systemctl show --property DefaultTasksMax
    cat /proc/sys/kernel/threads-max
    cat /proc/sys/kernel/pid_max

mistoll, 2020-10-01:
Your're right. We're running the node inside a systemd service. I wasn't aware of that, but there's a "Task" limit.
```
$ sudo systemctl status ros.service
● ros.service - ros
   Loaded: loaded (/etc/systemd/system/ros.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-10-01 14:25:40 UTC; 9min ago
   Main PID: 18499 (start_roslaunch)
   **Tasks: 63 (limit: 4444)**
```
I saw that same output with a high number of active Tasks when the problem was active. (Didn't copy the output.)

```
$ systemctl show --property DefaultTasksMax
DefaultTasksMax=4444
$ cat /proc/sys/kernel/threads-max
29632
$ cat /proc/sys/kernel/pid_max
32768
```

Now I think this has nothing to do with ROS at all. There are multiple nodes in the launch files and that could simply be to much for the 4444 Task limit. I'm going to split the service into two and see what's happening.

So I'm closing this for now. If there's any evidency, that there's a ros bug, I'll reopen the issue.

closed 2020-10-01 by mistoll