ros/ros_comm #2000: topic_tools relay support --filter flag like rostopic echo

AravindaDP, 2020-07-16:
I would like to request feature request on either `topic_tools relay` or `relay_field` tool to support `--filter` flag like `rostopic echo` tool does.

Idea is to forward only a selected subset of tf messages or anything like that on a different topic.

Alternative solutions I tried:
- I tried using `topic_tools transform` with a ternary operator but it still published an empty message for excluded case.
- rostopic pub with piped echo works but I need this solution to be able to run through a ros launch file.

dirk-thomas, 2020-07-16:
Please consider to contribute a pull request for your feature idea.

AravindaDP, 2020-07-16:
@dirk-thomas 

Thanks for the response. Sure I'll try submitting a PR

In the meantime I realized it's technically possible to wrap a shell script piping rotopic pub and echo for the same effect. But let's keep this opened for some time till either me or someone comes up with a PR as suggested.

peci1, 2021-11-22:
Just to save some time of others - here's a bash oneliner that achieves this behavior (although it's not very nice):

    rostopic pub /filtered_topic std_msgs/String < <(rostopic echo --filter 'm is not None' /in_topic)

closed 2025-05-31 by sloretz