ros/ros_comm #2040: Roslaunch does not work. machine.env_loader must have been assigned before...

ga74kud, 2020-09-10:
roslaunch ....
... logging to /home/hartmannadm/.ros/log/9032798e-f36e-11ea-8f73-40b0769e9890/roslaunch-insc02303-29320.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://127.0.0.1:35307/
remote[localhost-0] starting roslaunch
Value error: machine.env_loader must have been assigned before creating ssh child instance
The traceback for the exception was written to the log file

dirk-thomas, 2020-09-10:
Please provide reproducible steps of your problem.

dirk-thomas, 2020-10-05:
Closing due to no further response.

closed 2020-10-05 by dirk-thomas