ros/ros_comm #2043: free(): invalid next size (fast) in ros::TimerManager<>::threadFunc() ()

tylerjw, 2020-09-11:
```gdb
free(): invalid next size (fast)

Thread 24 "pick_bot_main" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fff77091700 (LWP 27810)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff69958b1 in __GI_abort () at abort.c:79
#2  0x00007ffff69de907 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff6b0bdfa "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff69e597a in malloc_printerr (str=str@entry=0x7ffff6b0da60 "free(): invalid next size (fast)") at malloc.c:5350
#4  0x00007ffff69ecfd0 in _int_free (have_lock=0, p=0x7fffb4002fa0, av=0x7fffb4000020) at malloc.c:4213
#5  __GI___libc_free (mem=0x7fffb4002fb0) at malloc.c:3124
#6  0x00007ffff7b01a66 in ros::TimerManager<ros::Time, ros::Duration, ros::TimerEvent>::threadFunc() () from /opt/ros/melodic/lib/libroscpp.so
#7  0x00007ffff5ef4bcd in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#8  0x00007ffff5ac66db in start_thread (arg=0x7fff77091700) at pthread_create.c:463
#9  0x00007ffff6a76a3f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

dirk-thomas, 2020-09-11:
Please provide reproducible steps of a minimal example.

tylerjw, 2020-09-11:
I now believe this was caused by memory corruption from somewhere else in the software.  I'm sorry about this, closing for now.

closed 2020-09-11 by tylerjw