ros/ros_comm #2048: roscore breaks after building the ros_comm package

Aniket-Gujarathi, 2020-09-21:
I cloned the ros_comm package in my workspace and built it. Then tried to run roscore but the system broke giving error:
`[FATAL] [1600667240.900393639]: ASSERTION FAILED
	file = /home/aniket/catkin_ws/src/ros_comm/clients/roscpp/include/ros/timer_manager.h
	line = 242
	cond = false
	message = 
[FATAL] [1600667240.900422273]: ros::TimerManager was instantiated by package roscpp, but neither BOOST_THREAD_HAS_CONDATTR_SET_CLOCK_MONOTONIC nor BOOST_THREAD_INTERNAL_CLOCK_IS_MONO is defined! Be aware that timers might misbehave when system time jumps, e.g. due to network time corrections.
[FATAL] [1600667240.900427469]: 
`
I am not able to understand what is causing this in ros_comm, but by removing ros_comm, roscore starts to work properly.
Please help.

dirk-thomas, 2020-09-22:
Please include more information about your platform, the versions you are using, how you have built or installed other ROS packages, the exact command you run etc. in order to allow others to help you.

Aniket-Gujarathi, 2020-09-29:
Hi, I changed the version of the package and then it began to run properly. I am not sure what was causing the issue earlier.

closed 2020-09-29 by Aniket-Gujarathi

maxlengdell, 2021-06-22:
I have the same issue atm. Could you be more specific with version and package. What and where did you change it?

FrankLin9981, 2021-06-24:
Hi,
Recently, I bump into the same issue. Does anyone have the solution without change the version of the package.
Thanks!

Initializing, 2023-01-01:
troubled me for 2 days with a full re-install of everything about ros, finally figured it out. a lot of ros tutorial does not seems to care about the version of ros_comm package. I'm running melodic on ubuntu 18.04 and the noetic version of ros_comm actually works totally fine before I start trying to work on turtlebot stuff. 

For the fix, remove the old ros_comm that does not match your ros version and install the exact one for your specific ros version. i.e., the ros_comm-melodic-devel for myself since I'm on ros melodic. 

Hope this can help some people.

ioarun, 2023-01-16:
If you're using ros noetic, try updating Boost version to >= 1.67. It worked for me with 1.81.0.