ros/ros_comm #2067: shutdown request always happens with using cached parameter if the same node name is used.

fujitatomoya, 2020-10-02:
### Problem

some node suddenly gets shutdown by master with "new node registered with same name". if the node uses cached parameter, it always leads to signal shutdown with "new node registered with same name". this might shut down the another node if [bumped_api](https://github.com/ros/ros_comm/blob/f06d584965a80ed6a966456758de20d1a63b6893/tools/rosmaster/src/rosmaster/registrations.py#L465) matches.

### Reproducible Procedure (confirmation that "new node registered with same name" happens)

with ros::noetic,

```
> cd <catkin_workspace>/src
> git clone -b 20201002-shutdown-request-debug git@github.com:fujitatomoya/ros_comm.git 
> cd <catkin_workspace>
> catkin_make install
> source devel/setup.bash
> roscore // Terminal-1
> rosrun roscpp_tutorials cached_parameters // Terminal-2 1st time
> rosrun roscpp_tutorials cached_parameters // Terminal-2 2nd time
> cat ~/.ros/log/latest/master.log
```
you can see the following debug information during `2nd time` of `rosrun roscpp_tutorials cached_parameters`,

```
[rosmaster.master][INFO] 2020-10-02 04:46:50,537: _register_node_api: /cached_parameters, http://9b29c6add660:45073/, http://9b29c6add660:45715/
[rosmaster.master][INFO] 2020-10-02 04:46:50,537: /my_cached_param/
[rosmaster.master][INFO] 2020-10-02 04:46:50,537: /rosout_disable_topics_generation/
```

### Description

it means that cachedParameter `/my_cached_param/` and `/rosout_disable_topics_generation/` are still in the master registry. So that if node with the same name comes online, master detects the same node name and sends shutdown request to the previous node. at this moment, if the port number is used by another node (which is running with different node name) gets suddenly shutdown.

fujitatomoya, 2020-10-02:
I believe the following should be addressed,

- call `unsubscribeParam` with parameter name when parameter is deleted. (cached parameter in user application)
- call `unsubscribeParam` to unregist all the cached parameter when shutdown. (rosout_disable_topics_generation)

fujitatomoya, 2020-10-02:
could be related to https://github.com/ros/ros_comm/issues/2018

petioptrv, 2020-11-02:
@dirk-thomas is this a ROS-beginner friendly issue? If so, and if no work is underway yet, I'd like to give it a shot at addressing it.

dirk-thomas, 2020-11-02:
@petioptrv I am not a maintainer of this repo anymore. Maybe one of the current maintainers can comment: @jacobperron, @mjcarroll, @sloretz.

sloretz, 2020-11-02:
@fujitatomoya Did #2068 resolve this issue?

fujitatomoya, 2020-11-03:
@sloretz 

yes! sorry about that, i will close this issue.

closed 2020-11-03 by fujitatomoya