ros/ros_comm #2071: roslaunch: eval does more than it tells it allows

peci1, 2020-10-06:
The documentation of `_eval_dict` is not verbose enough, but it creates the impression in me that `$(eval)` in roslaunch XML should only allow math functions, 'list', 'dict', 'map', 'str', 'float', 'int' and nothing more.

https://github.com/ros/ros_comm/blob/089dd64a6c2d7769d2d8d3b10808eec621a3acf3/tools/roslaunch/src/roslaunch/substitution_args.py#L298-L311

Consider this simple launch file:

```
<launch>
  <param name="a" value="$(eval filter(lambda x: x > 1, [1, 2]))" />
</launch>
```

After launching it, we see:

```
PARAMETERS
 * /a: [2]
...
```

Apparently, `eval` also allows `filter`. The same holds for other unlisted builtin symbols like `min`, `max` etc.

So there is some leak of the global builtins into Python's `eval`. It is here:

https://github.com/ros/ros_comm/blob/089dd64a6c2d7769d2d8d3b10808eec621a3acf3/tools/roslaunch/src/roslaunch/substitution_args.py#L347

From [`eval` documentation](https://docs.python.org/3/library/functions.html#eval):

>  eval(expression[, globals[, locals]])

>    The arguments are a string and optional globals and locals. If provided, globals must be a dictionary. If provided, locals can be any mapping object.

>    The expression argument is parsed and evaluated as a Python expression (technically speaking, a condition list) using the globals and locals dictionaries as global and local namespace. If the globals dictionary is present and does not contain a value for the key \_\_builtins\_\_, a reference to the dictionary of the built-in module builtins is inserted under that key before expression is parsed. This means that expression normally has full access to the standard builtins module and restricted environments are propagated.

However, roslaunch only passes `{}` as `globals`, which means it does not prevent passing the standard builtins (contrary to the documentation comment above). [Roslaunch wiki](http://wiki.ros.org/roslaunch/XML) doesn't help regarding the desired behavior.

I can imagine that on one side having more builtins accessible is a good thing, but things like `$(eval str(open('/dev/zero', 'r')))` should definitely not be allowed (`$ rosparam get /a` currently returns `<open file '/dev/zero', mode 'r' at 0x7f74652d7ed0>`).

The fix is easy: pass `{'__builtins__': {}}` as the `globals` argument. `eval` would then really allow only the listed functions. Or even better - `_eval_list` could be passed as `globals` and context args as `locals`, which would allow to get rid of `_DictWrapper` and even allow introduction of local variables (that would e.g. resolve #2056).

I can prepare a PR which would fix this. However, as a lot of stuff was allowed with the current behavior, it would definiely break downstream code. For example our launch files are full of `$(eval len(a))`. So the PR would have to also pick the broadest possible "safe" subset of the default builtins which should be allowed. I believe nobody needs to call `open()` in `eval`, for example, so this is one candidate to forbid. And there are much more. Should we try to curate a list of the allowed builtins? It could then also be added to the wiki page so that people know exactly what they can use.

peci1, 2020-10-06:
<param name="a" value="$(eval eval('[x for x in [1,2] if x == 1]', {}, {}))" />

:-D

florcabral, 2023-07-12:
Returning to this issue, I was wondering if there will be a CVE assigned for this vulnerability?

peci1, 2023-07-12:
I'm not really sure this is a vulnerability itself. It would need user code to pass commandline arguments to the eval, which is obviously generally not a very good idea. So I guess vulnerabilities could only be seen in downstream code wrongly using eval.