ros/ros_comm #2083: No way to get server object of already registered service once the handler is not accessible

furushchev, 2020-10-22:
After initializing `rospy.Server` object with one callback registered, if the server object is not accessible, there is no way to get an object to the servicce of the name.

```python
#!/usr/bin/env python
# test_service.py

import rospy
from std_srvs.srv import Empty, EmptyResponse

def func():
    s = rospy.Service('~foo', Empty, lambda req: EmptyResponse())

rospy.init_node('test_service')

func()
func()
```

This causes an error:

```
Traceback (most recent call last):
  File "test_service.py", line 15, in <module>
    func()
  File "test_service.py", line 9, in func
    s = rospy.Service('~foo', Empty, lambda req: EmptyResponse())
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 721, in __init__
    get_service_manager().register(self.resolved_name, self)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/service.py", line 116, in register
    raise ServiceException(err)
rospy.service.ServiceException: service [/test_service/foo] already registered
```

The error can be avoided by chainging the `func` like below:

```python
def func():
    s = rospy.Service('~foo', Empty, lambda req: EmptyResponse())
    s.shutdown()
```

The problem is, in the case we must change the `func` and ensure the `shutdown` method is called.
There is no way to initialize service with the same name if the first `rospy.Service` object is no more accessible.

Does anyone have idea to workaround this issue or any advice to fix the problem?
Thank you very much in advance!

furushchev, 2020-10-22:
Would it be good to add an optional argument `overwrite` to `rospy.Service` that allows overwriting callback registration for the named service?

```python
import rospy
from std_srvs.srv import Empty, EmptyResponse

rospy.init_node('test_service')

s1 = rospy.Service('~foo', Empty, lambda req: EmptyResponse())
s2 = rospy.Service('~foo', Empty, lambda req: EmptyResponse(), overwrite=True)
```

sloretz, 2020-10-23:
Thanks for opening an issue!

> manager = rospy.impl.registration.get_service_manager()
> s = manager.get_service('{}/foo'.format(rospy.get_name())

It looks like that would get an instance of [`rospy.service._Service`](https://github.com/ros/ros_comm/blob/090daa56f57d18c39946f107bcee5e3efefb2cac/clients/rospy/src/rospy/service.py#L54), which seems to be different from [`rospy.Service`](https://github.com/ros/ros_comm/blob/090daa56f57d18c39946f107bcee5e3efefb2cac/clients/rospy/src/rospy/__init__.py#L72) AKA [`rospy.impl.tcpros_service.Service`](https://github.com/ros/ros_comm/blob/090daa56f57d18c39946f107bcee5e3efefb2cac/clients/rospy/src/rospy/impl/tcpros_service.py#L680)

> s2 = rospy.Service('~foo', Empty, lambda req: EmptyResponse(), overwrite=True)

How would this signal to the original `rospy.Service` instance that it is invalid? What happens if the original service object is still in use, and methods are called on it? Say `serv2` is created with `overwrite=True`, but at the same time [`serv1.shutdown()`](https://github.com/ros/ros_comm/blob/090daa56f57d18c39946f107bcee5e3efefb2cac/clients/rospy/src/rospy/impl/tcpros_service.py#L583) is being called?