ros/ros_comm #2085: ros::Timer won't start if initialized before dynamic reconfigure server

klaxalk, 2020-10-28:
Hey, I am posting this issue since I haven't got any answers here: https://answers.ros.org/question/364115/rostimer-wont-start-if-initialized-before-dynamic-reconfigure-server/. But I am not sure if this is the right place. Forgive me if I am wrong.

## The problem

My **ros::Timer** won't start if I initialize it before the initialization of a **dynamic reconfigure server**.

## Environment

* Ubuntu 20.04 + ROS Noetic
* empty workspace with no specific flags
* tested on amd64 or arm64

## Observations

* when **ros::Timer** is initialized before a **dynamic reconfigure server** (DRS), the timer won't start
* it affects fast timers (100 Hz and more) rather than slow timers
* it is non-deterministic, sometimes it starts correctly
* the higher the number of DRS parameters, the higher the chance of replicating the issue (10+ parameters = near 100% chance)
* seems not to be influenced by the computational resources, happens on i9-9900K as well as on Rpi4
* does not happen at all on 18.04 + ROS Melodic

## Minimal non-working example

* https://github.com/klaxalk/ros_timer_drs_bug
* has two timers, 1 Hz and 100 Hz
* has a pre-compiler `#define` which can switch the order of initialization
* the 100 Hz timer won't run

## Solution

* I don't know, the DRS seems to be devoid of ros::Timers.
* Can't say for sure, but swapping the order is not a viable workaround for ros::Pluginlib plugins. So far it looks like some plugins' timers are blocked by other plugins' DRSs.
* the only workaround I found is to use a slow timer (or a thread) to check the activity and restart any broken fast timers... but... duh...

## Sources

Full sources here: https://github.com/klaxalk/ros_timer_drs_bug

The main cpp file:
```cpp
#include <ros/ros.h>
#include <nodelet/nodelet.h>

#include <dynamic_reconfigure/server.h>
#include <timer_tester/timer_testerConfig.h>

#define TIMERS_BEFORE 1

namespace timer_tester
{

class TimerTester : public nodelet::Nodelet {

public:
  virtual void onInit();

private:
  ros::NodeHandle nh_;

  // | --------------- dynamic reconfigure server --------------- |

  boost::recursive_mutex                           mutex_drs_;
  typedef timer_tester::timer_testerConfig         DrsConfig_t;
  typedef dynamic_reconfigure::Server<DrsConfig_t> Drs_t;
  boost::shared_ptr<Drs_t>                         drs_;
  void                                             callbackDrs(timer_tester::timer_testerConfig& config, uint32_t level);

  // | ------------------------- timers ------------------------- |

  ros::Timer timer_fast_;
  void       timerFast(const ros::TimerEvent& te);

  ros::Timer timer_slow_;
  void       timerSlow(const ros::TimerEvent& te);
};

void TimerTester::onInit() {

  ros::NodeHandle nh_ = nodelet::Nodelet::getMTPrivateNodeHandle();

  ros::Time::waitForValid();

  ROS_INFO("[TimerTester]: initializing");

#if TIMERS_BEFORE == 1
  ROS_INFO("[TimerTester]: creating timers before DRS");
  timer_fast_ = nh_.createTimer(ros::Rate(100.0), &TimerTester::timerFast, this);
  timer_slow_ = nh_.createTimer(ros::Rate(1.0), &TimerTester::timerSlow, this);
#endif

  // | --------------- dynamic reconfigure server --------------- |

  drs_.reset(new Drs_t(mutex_drs_, nh_));
  Drs_t::CallbackType f = boost::bind(&TimerTester::callbackDrs, this, _1, _2);
  drs_->setCallback(f);

#if TIMERS_BEFORE == 0
  ROS_INFO("[TimerTester]: creating timers after DRS");
  timer_fast_ = nh_.createTimer(ros::Rate(100.0), &TimerTester::timerFast, this);
  timer_slow_ = nh_.createTimer(ros::Rate(1.0), &TimerTester::timerSlow, this);
#endif

  ROS_INFO_ONCE("[TimerTester]: initialized");
}

// | --------------------- timer callbacks -------------------- |

void TimerTester::timerFast([[maybe_unused]] const ros::TimerEvent& te) {

  ROS_INFO_THROTTLE(0.1, "[TimerTester]: 100 Hz timer spinning");
}

void TimerTester::timerSlow([[maybe_unused]] const ros::TimerEvent& te) {

  ROS_INFO_THROTTLE(0.1, "[TimerTester]: 1 Hz timer spinning");
}

void TimerTester::callbackDrs([[maybe_unused]] timer_tester::timer_testerConfig& config, [[maybe_unused]] uint32_t level) {

  ROS_INFO("[TimerTester]: callbackDrs() called");
}

}  // namespace timer_tester

#include <pluginlib/class_list_macros.h>
PLUGINLIB_EXPORT_CLASS(timer_tester::TimerTester, nodelet::Nodelet);
```

The dynamic reconfigure server config:
```python
#!/usr/bin/env python

PACKAGE = "timer_tester"
import roslib;
roslib.load_manifest(PACKAGE)

from dynamic_reconfigure.parameter_generator_catkin import *

gen = ParameterGenerator()

main = gen.add_group("Main");
main.add("a", double_t, 1, "A", 1.0, 0.0, 10.0);
main.add("b", double_t, 2, "B", 1.0, 0.0, 10.0);
main.add("c", double_t, 4, "C", 1.0, 0.0, 10.0);
main.add("d", double_t, 8, "D", 1.0, 0.0, 10.0);
main.add("e", double_t, 16, "E", 1.0, 0.0, 10.0);
main.add("f", double_t, 32, "F", 1.0, 0.0, 10.0);
main.add("g", double_t, 64, "G", 1.0, 0.0, 10.0);
main.add("h", double_t, 128, "H", 1.0, 0.0, 10.0);
main.add("i", double_t, 256, "I", 1.0, 0.0, 10.0);
main.add("j", double_t, 512, "J", 1.0, 0.0, 10.0);
main.add("k", double_t, 1024, "K", 1.0, 0.0, 10.0);
main.add("l", double_t, 2048, "L", 1.0, 0.0, 10.0);

exit(gen.generate(PACKAGE, "TimerTester", "timer_tester"))
```

klaxalk, 2020-11-29:
Ok, after building my own Timer library to work around this, I found out that the DRS affects subscribers as well. Similarly, subscribers for fast topics (>= 100 Hz) can freeze completely when DRS is present. Again, the more params the DRS has the worse it is. I will create a _breaking example_ for the subscribers as well and probably resubmit this issue to [https://github.com/ros/dynamic_reconfigure](https://github.com/ros/dynamic_reconfigure].

klaxalk, 2020-11-29:
I finally got it. The problem is not with DRS in particular but with anything that takes a non-zero amount of time to execute. It can be broken with sleep() or just with some traditional operations. The problem is in fact caused by the **nodelet**, specifically by the first commit that got to Noetic: https://github.com/ros/nodelet_core/commit/5272c3462e7c7ebab9848ed943d5c926b1e75842. The commit was intended to fix this particular issue, but it in fact causes it. They disable callback queues before calling `onInit()` and re-enable them after. But it seems that this is not a good solution. I will submit an issue there. In the meantime, a temporary workaround is to compile the nodelet without that commit.

closed 2020-11-30 by klaxalk

BruceChanJianLe, 2021-05-07:
Hi there, may I know if the commit is merged to melodic branch as well? It appears that melodic has similar issue of timer not running when dynamic reconfigure server is declared before it.

klaxalk, 2021-05-07:
Hey, I don't have melodic around anymore and I don't see any melodic-specific branch in the repo. Could you pls check what version of nodelet_core do you have?

```bash
cat /opt/ros/melodic/share/nodelet_core/package.xml
```

BruceChanJianLe, 2021-05-07:
Hi, thanks for getting back, here is the requested file.
```xml
<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format2.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="2">
  <name>nodelet_core</name>
  <version>1.9.16</version>
  <description>Nodelet Core Metapackage</description>
  <maintainer email="mikael@osrfoundation.org">Mikael Arguedas</maintainer>
  <license>BSD</license>

  <url type="website">http://www.ros.org/wiki/nodelet_core</url>
  <url type="bugtracker">https://github.com/ros/nodelet_core/issues</url>
  <url type="repository">https://github.com/ros/nodelet_core</url>

  <author>Tully Foote</author>

  <buildtool_depend>catkin</buildtool_depend>

  <!-- Old stack contents -->
  <exec_depend>nodelet</exec_depend>
  <exec_depend>nodelet_topic_tools</exec_depend>

  <export>
    <metapackage/>
  </export>

</package>
```

klaxalk, 2021-05-07:
Hey, as far as I know, that should still be fine, meaning, before they broke it. If you want to crosscheck if your problem really comes from this issue, clone my fork of https://github.com/ctu-mrs/nodelet_core/tree/callback_fix and compile it in your workspace. It should print, so you can verify, the patched version is there.

[ INFO] [1620377125.241694759, 9.188000000]: **(patched)** Done initing nodelet <nodelet_name>