ros/ros_comm #2100: In XmlRpcSocket.cpp,  missing matching closing curly bracket }

aakash641, 2020-12-01:
In XmlRpcSocket.cpp, a closing } corresponding to XmlRpcSocket::connect function is missing.
That is, if you search for a closing } for the bracket { in line 218, you won't find. 

Please help.

fujitatomoya, 2020-12-01:
> bracket { in line 218,

are you saying this one does not have closing?

https://github.com/ros/ros_comm/blob/090daa56f57d18c39946f107bcee5e3efefb2cac/utilities/xmlrpcpp/src/XmlRpcSocket.cpp#L217

i believe that closure is here,

https://github.com/ros/ros_comm/blob/090daa56f57d18c39946f107bcee5e3efefb2cac/utilities/xmlrpcpp/src/XmlRpcSocket.cpp#L220

are you seeing any build failure because of this?

aakash641, 2020-12-01:
I am saying that the below does not have a closing:

https://github.com/ros/ros_comm/blob/090daa56f57d18c39946f107bcee5e3efefb2cac/utilities/xmlrpcpp/src/XmlRpcSocket.cpp#L201

Surprisingly, there is no build failure because of this.

aakash641, 2020-12-01:
Also, for this line too: (total 2 { for which there is NO closing }):--

https://github.com/ros/ros_comm/blob/090daa56f57d18c39946f107bcee5e3efefb2cac/utilities/xmlrpcpp/src/XmlRpcSocket.cpp#L211

fujitatomoya, 2020-12-01:
I think that it is because of `ifdef` statement that makes it look complicated, but both have closing.

aakash641, 2020-12-01:
I am sorry, but I disagree. 
If both have closing, could you please provide the hyperlink to both closings ?
I think you are not checking properly. I have considered the ifdef statements too.

fujitatomoya, 2020-12-01:
> If both have closing, could you please provide the hyperlink to both closings ?

https://github.com/ros/ros_comm/blob/090daa56f57d18c39946f107bcee5e3efefb2cac/utilities/xmlrpcpp/src/XmlRpcSocket.cpp#L201

to

https://github.com/ros/ros_comm/blob/090daa56f57d18c39946f107bcee5e3efefb2cac/utilities/xmlrpcpp/src/XmlRpcSocket.cpp#L287

and,

https://github.com/ros/ros_comm/blob/090daa56f57d18c39946f107bcee5e3efefb2cac/utilities/xmlrpcpp/src/XmlRpcSocket.cpp#L211

to

https://github.com/ros/ros_comm/blob/090daa56f57d18c39946f107bcee5e3efefb2cac/utilities/xmlrpcpp/src/XmlRpcSocket.cpp#L222

> I have considered the ifdef statements too.

if that so, would you mind sharing the fix as Pull Request? i think that would be nice to see the exact code where needs to be fixed.

aakash641, 2020-12-02:
You are saying that { in line 211 closes at line 222.
But actually } in Line 222 is closure of { in Line 215.

I request you to please check properly instead of spamming this page.

https://github.com/ros/ros_comm/blob/090daa56f57d18c39946f107bcee5e3efefb2cac/utilities/xmlrpcpp/src/XmlRpcSocket.cpp#L222

is opened at

https://github.com/ros/ros_comm/blob/090daa56f57d18c39946f107bcee5e3efefb2cac/utilities/xmlrpcpp/src/XmlRpcSocket.cpp#L215

not at line 211

aakash641, 2020-12-02:
I don't know how to share the fix as Pull Request. Could you please guide me ?

fujitatomoya, 2020-12-02:
CC: @sloretz @jacobperron @mjcarroll 

> I request you to please check properly instead of spamming this page.

this is not constructive...cz i do not think i am making trouble here, just trying to understand. i am just telling that what i can see. i guess that best way is to see the fix, could you consider the PR?

aakash641, 2020-12-02:
Ok, sorry. I felt it is straight-forward clearly visible thing. 
I take back my words. 
After giving it a thought, I am doubtful whether you are checking the noetic-devel branch ? 

> could you consider the PR?

I have no idea how to do it. If you or anyone can guide me, I will be happy to contribute.

aakash641, 2020-12-02:
@fujitatomoya Coule you please share the complete URL of the code that you are looking at ?

I have raised the issue in the following code:

https://github.com/ros/ros_comm/blob/noetic-devel/utilities/xmlrpcpp/src/XmlRpcSocket.cpp

sloretz, 2020-12-02:
Let's have a machine do the work. First building `xmlrpcpp` succeeds, so it's very unlikely that the curly brackets are mismatched. To make sure, I built with the cmake argument `-DCMAKE_VERBOSE_MAKEFILE=True` and noted the compile command for `XmlRpcSocket.cpp`.

```console
/usr/bin/c++  -DROS_BUILD_SHARED_LIBS=1 -Dxmlrpcpp_EXPORTS -I/home/sloretz/bigssd/noetic/src/ros_comm/utilities/xmlrpcpp/include -I/home/sloretz/bigssd/noetic/src/ros_comm/utilities/xmlrpcpp/libb64/include -I/home/sloretz/bigssd/noetic/install/rostime/include -I/home/sloretz/bigssd/noetic/install/cpp_common/include  -fPIC   -Wall -Wextra -o CMakeFiles/xmlrpcpp.dir/src/XmlRpcSocket.cpp.o -c /home/sloretz/bigssd/noetic/src/ros_comm/utilities/xmlrpcpp/src/XmlRpcSocket.cpp
```

Then I `cd`'d to the build directory, replaced `-o CMakeFiles/xmlrpcpp.dir/src/XmlRpcSocket.cpp.o` with `-E` to get the output after preprocessing, and saved that in a file `/tmp/post_processed.txt`. That gives me the source file with all `#ifdef` statements resolved, at least as they are on Ubuntu Focal.

```
/usr/bin/c++  -DROS_BUILD_SHARED_LIBS=1 -Dxmlrpcpp_EXPORTS -I/home/sloretz/bigssd/noetic/src/ros_comm/utilities/xmlrpcpp/include -I/home/sloretz/bigssd/noetic/src/ros_comm/utilities/xmlrpcpp/libb64/include -I/home/sloretz/bigssd/noetic/install/rostime/include -I/home/sloretz/bigssd/noetic/install/cpp_common/include  -fPIC   -Wall -Wextra -E -c /home/sloretz/bigssd/noetic/src/ros_comm/utilities/xmlrpcpp/src/XmlRpcSocket.cpp > /tmp/post_processed.txt
```

Finally I opened the file in vim and used `:s/{//gn` and `:s/}//gn` to count the number of `{` and `}`. Those commands said 2120 `{` and 2120 `}`, so there's an equal number of them. Since there's the same number of matching curly braces and it successfully compiles, it seems reasonable that all braces are matched.

Everything looks fine to me, so I'll close this issue for now. If you still think there's a bug then please [open a pull request](https://docs.github.com/en/enterprise/2.16/user/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request-from-a-fork).

closed 2020-12-02 by sloretz