ros/ros_comm #2106: rospy uses depracated warn() instead of warning() (important; with patch)

mikolajz, 2020-12-16:
In Ubuntu 20.04 with ROS Noetic, running code using `rospy` results in a lot of warnings like:
```
/opt/ros/noetic/lib/python3/dist-packages/rospy/core.py:183: DeprecationWarning: The 'warn' method is deprecated, use 'warning' instead
    logfunc(msg, *args, **kwargs)
```

The attached patch fixes it.

The surprising part is that this is important for us, as when we run our `pytest` in Ubuntu 20.04, we get more such warnings than the default scrolling buffer of the terminal, thus we don't see the other messages above.

[rospy.diff.txt](https://github.com/ros/ros_comm/files/5704492/rospy.diff.txt)

jacobperron, 2020-12-23:
@mikolajz Thanks for the report and patch file! Would you be willing to open a pull request directly?

mikolajz, 2021-01-25:
Sorry for the delay, this got lost in the e-mails around Christmas, but now I've created the PR https://github.com/ros/ros_comm/pull/2120 .

jacobperron, 2021-02-01:
Resolved by #2120

closed 2021-02-01 by jacobperron

mikolajz, 2021-02-02:
Thanks.

What is the release schedule for ROS? Can we expect this patch to be released in DEB packages for ROS Noetic, or will it be part of some ROS Oxxx?

I'm asking since we have a patch in our local Docker image and wonder when we should prepare for it not to apply cleanly.

jacobperron, 2021-02-11:
@mikolajz You can expect this patch in an update to Noetic.

It's hard to give a timeline for the next release, but I can try and at least make a source release and get it into testing sometime next week.