ros/ros_comm #2109: view size question

NeilZhy, 2020-12-21:
https://github.com/ros/ros_comm/blob/a825290daf5f86d225efda4fd3011dc196438a91/tools/rosbag_storage/src/view.cpp#L228
I took a look at the internal implementation, but I don’t understand why “update” is called. I feel it’s an extra operation.

sloretz, 2020-12-29:
Maybe @tfoote the original author would know, though it may be hard to remember since it looks like it was written 10 years ago

sloretz, 2021-01-11:
It seems unlikely that this question will get answered, so I'll close it. For more questions I recommend asking on https://answers.ros.org/questions/ . If you remove the `update` call and testing show's it was unnecessary, please open a pull request removing it.

closed 2021-01-11 by sloretz