ros/ros_comm #2112: TCPROS ERROR

ManChrys, 2020-12-31:
Hello guys!

First of all Merry Christmas and i wish hapiness to all of you guys!!

So the reason im writting here is that i have the following error : 

**~/catkin_ws/src/hdl_graph_slam/rviz$ rviz -d hdl_graph_sm.rviz**

[ INFO] [1609337372.505558332]: rviz version 1.13.15
[ INFO] [1609337372.505604409]: compiled against Qt version 5.9.5
[ INFO] [1609337372.505612675]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1609337372.510855639]: Forcing OpenGl version 0.
[ INFO] [1609337373.062548537]: Stereo is NOT SUPPORTED
[ INFO] [1609337373.062603211]: OpenGl version: 4,6 (GLSL 4,6).
[ERROR] [1609339787.830933100]: a message of over a gigabyte was predicted in tcpros. that seems highly unlikely, so I'll assume protocol synchronization is lost.

Im using ros-melodic on Ubuntu 18.04. Im running the following launch file : 

[launch.txt](https://github.com/ros/ros_comm/files/5756664/launch.txt)

Im using this launch file to build my bag after it's recording

Any help will be useful.

Thanks in advance guys!!

fujitatomoya, 2021-01-05:
it seems that it tried to transmit really huge data which is bigger than 1GB, https://github.com/ros/ros_comm/blob/f4ab23fe6d9e64fcc2d08cbd88ba368f5188b131/clients/roscpp/src/libros/transport_publisher_link.cpp#L174-L182

Are you actually trying to send the data over 1GB? could you check the data size of a message?

ManChrys, 2021-01-05:
Hello and happy new year @fujitatomoya !!! Thanks for the response.

im not sure if i used the right command in order to answer your question so : 

[image]

[image]

Generally im using "rosbag play bagname.bag" to  create the frames and after that im using the created frames to interactive_slam. Τhis is about the process.

im seeing that im over 1gb but that happened when i have to "rosbag play bagname.bag" , because i have big bag files.

How can solve this ? Any suggestion ? Or i have to split bag files to smaller files.

As you can see im relatively new on ros so be patient :D 

Thanks in adnance.

fujitatomoya, 2021-01-05:
> im seeing that im over 1gb

correct, it is over 1GB as maximum. in that case on the subscription, it will print the error message and drop the connection since this is expected to be really unlikely to happen.

what i would try is,
1. patch and increase the threshold to 2GB, which is currently set 1GB.

```
diff --git a/clients/roscpp/src/libros/transport_publisher_link.cpp b/clients/roscpp/src/libros/transport_publisher_link.cpp
index 82bd1400f..319cdfff4 100644
--- a/clients/roscpp/src/libros/transport_publisher_link.cpp
+++ b/clients/roscpp/src/libros/transport_publisher_link.cpp
@@ -171,7 +171,7 @@ void TransportPublisherLink::onMessageLength(const ConnectionPtr& conn, const bo

   uint32_t len = *((uint32_t*)buffer.get());

-  if (len > 1000000000)
+  if (len > 2000000000)
   {
     ROS_ERROR("a message of over a gigabyte was " \
                 "predicted in tcpros. that seems highly " \
```

2. build patched `ros_comm` meta-package with catkin workspace. see http://wiki.ros.org/catkin/Tutorials

ManChrys, 2021-01-07:
@fujitatomoya thanks for the help. I'll try it and i'll inform you next week.

Thanks again :)

closed 2021-05-06 by ManChrys