ros/ros_comm #2119: publisher does not work on multiprocessing mudule

ye12121, 2021-01-15:
Hi,

I need to publish data by using multiprocessing in my situation. it worked for following code, and i saw the data by rostopic echo /test

`import rospy
from std_msgs.msg import String
rospy.init_node('t', anonymous=True)

def pub(hz=20):
    pub = rospy.Publisher('/test', String, queue_size=10)
    rate = rospy.Rate(hz)
    while not rospy.is_shutdown():
        pub.publish('*'*10)
        rate.sleep()

if __name__ == '__main__':
    pub()
`

but when i use Process module to ran it, like:

`import rospy
from std_msgs.msg import String
from multiprocessing import Process
rospy.init_node('t', anonymous=True)

def pub(hz=20):
    pub = rospy.Publisher('/test', String, queue_size=10)
    rate = rospy.Rate(hz)
    while not rospy.is_shutdown():
        pub.publish('*'*10)
        rate.sleep()

if __name__ == '__main__':
    p = Process(target=pub, args=(20, ))
    p.start()
    # pub()
`
and there is no message i saw.
Has anyone faced this problem before?

mikepurvis, 2021-01-15:
Is it any different if you `init_node` inside the process function? Either way, it's not super surprising that this doesn't work as expected; rospy's threading model is not well defined.

I would say that if you're trying to do the kind of task where this setup is needed, it may be best to either rewrite in roscpp, or if it _must_ be Python, keep the ROS API on the main thread, and use multiprocessing-aware queues and so on to communicate work out to your worker processes.

fujitatomoya, 2021-01-19:
> Is it any different if you init_node inside the process function? Either way, it's not super surprising that this doesn't work as expected; rospy's threading model is not well defined.

:+1: 

`rospy.init_node` needs to be called in process space. so the following works okay. (confirmed with `noetic-devel`)

```
import rospy
from std_msgs.msg import String
from multiprocessing import Process

def func(hz=20):
    pub = rospy.Publisher('/test', String, queue_size=10)
    rospy.init_node('t', anonymous=True) # HERE HERE HERE
    rate = rospy.Rate(hz)
    while not rospy.is_shutdown():
        pub.publish('*'*10)
        rate.sleep()

if __name__ == '__main__':
    p = Process(target=func, args=(20, ))
    p.start()
    # func()
```

i do not see this is the problem.

ye12121, 2021-01-20:
It did word. thanks mikepurvis, fujitatomoya.

closed 2021-01-20 by ye12121