ros/ros_comm #2123: Function recv_ Buff () when the remote end is closed, resulting in abnormal reconnection.

jinmenglei, 2021-01-30:
When I use rosbridge, after I send a message, I close the client connected through rosbridge, which will result in "d= sock.recv (buff_size)"  After receiving an empty string.
[image] 
An exception is thrown, and then the socket is reconnected. Finally, the waiting time exceeds 32S. It is difficult to connect, and the port will be wasted when reconnecting. I refer to the C + + code and find that the processing logic is different.
[image]
 C + + receives zero and closes the connection directly instead of reconnecting. Can we consider closing it, like this:
[image]

The test is OK after I modify it

fujitatomoya, 2021-02-01:
this sounds reasonable to me. since transport is terminated with 'TransportTerminated' we would not want to reconnect. (in roscpp, if that is neither `EAGAIN` nor `EWOULDBLOCK`, it is closed and done.)

> if str(e) == "unable to receive data from sender, check sender's logs for details":

instead of that, i would add except catch statement `TransportTerminated` before `TransportException`.

jinmenglei, 2021-02-01:
@fujitatomoya Thank you. After you fix it, I will pull the code again. I like ROS and hope that the UDP part of ros1 can be updated later.

commit b3f8e3c7 (2021-09-01): Fix #2123:  Do not raise exception if socket is busy in TCPROSTransport (#2131)

Make behavior consistent with the C++ implementation.
commit 8a35468b (2022-01-26): Fix #2123:  Do not raise exception if socket is busy in TCPROSTransport (#2131)

Make behavior consistent with the C++ implementation.
closed 2021-09-01 by jacobperron