ros/ros_comm #2125: Inconsistient behaviour between multiple latched publishers & subscribers on the same node vs different nodes

stevegolton, 2021-02-02:
#1544 fixes the issue where multiple latched publishers in the same node overwrite each other's latched message. This works fine for subscribers from external nodes, but subscribers on the same node still only receive the most recently published latched message.

Example cpp node:
```cpp
#include <ros/ros.h>
#include <std_msgs/String.h>

void latched_pubs()
{
    ros::NodeHandle nh;

    auto pub0 = nh.advertise<std_msgs::String>("latched", 100, true);
    auto pub1 = nh.advertise<std_msgs::String>("latched", 100, true);

    std_msgs::String msg;
    msg.data = "hi0";
    pub0.publish(msg);

    msg.data = "hi1";
    pub1.publish(msg);

    auto sub = nh.subscribe<std_msgs::String>("latched", 1, [](std_msgs::StringConstPtr msg){
        ROS_INFO("Got a msg: %s", msg->data.c_str());
    });

    ros::spin();
}
```
Example output:
```bash
[ INFO] [1612254563.074192987]: Got a msg: hi1
```

Example output from another node:
```bash
$ rostopic echo /latched
data: "hi0"
---
data: "hi1"
---
```

I realize this might be a nitpick but the fix in #1544, while really useful, makes this behavior inconsistent.

jinmenglei, 2021-02-02:
@stevegolton you can change `queue_size` from 1 to 2
```
ros::Subscriber sub = handle.subscribe("my_topic", 1, &Foo::callback, &foo_object);
\endverbatim
   *
   * \param M [template] M here is the callback parameter type (e.g. const boost::shared_ptr<M const>& or const M&), \b not the message type, and should almost always be deduced
   * \param topic Topic to subscribe to
   * \param queue_size Number of incoming messages to queue up for
   * processing (messages in excess of this queue capacity will be
   * discarded).
   * \param fp Member function pointer to call when a message has arrived
   * \param obj Object to call fp on
   * \param transport_hints a TransportHints structure which defines various transport-related options
   * \return On success, a Subscriber that, when all copies of it go out of scope, will unsubscribe from this topic.
   * On failure, an empty Subscriber which can be checked with:
\verbatim
ros::NodeHandle nodeHandle;
void Foo::callback(const std_msgs::Empty::ConstPtr& message) {}
boost::shared_ptr<Foo> foo_object(boost::make_shared<Foo>());
ros::Subscriber sub = nodeHandle.subscribe("my_topic", 1, &Foo::callback, foo_object);
```

stevegolton, 2021-02-02:
@jinmenglei Well, I feel foolish. Even changing the size to >=2 without reordering the subscribe seems to work for me. Thanks

```
void latched_pubs()
{
    ros::NodeHandle nh;

    auto pub0 = nh.advertise<std_msgs::String>("latched", 100, true);
    auto pub1 = nh.advertise<std_msgs::String>("latched", 100, true);

    std_msgs::String msg;
    msg.data = "hi0";
    pub0.publish(msg);

    msg.data = "hi1";
    pub1.publish(msg);

    auto sub = nh.subscribe<std_msgs::String>("latched", 2, [](std_msgs::StringConstPtr msg){
        ROS_INFO("Got a msg: %s", msg->data.c_str());
    });

    ros::spin();
}

int main(int argc, char *argv[])
{
    ros::init(argc, argv, "multiple_latched_pubs");

    latched_pubs();

    return 0;
}
```

closed 2021-02-02 by stevegolton