ros/ros_comm #2133: rosbag and cryptodome

flabrosse, 2021-02-16:
Currently making packages for gentoo.  On this distribution, pycryptodome is used, but this creates cryto packages in the site-packages structure.  This means that bag.py needs to be patched with something like:
```
diff --git a/tools/rosbag/src/rosbag/bag.py b/tools/rosbag/src/rosbag/bag.py
index 24c3e054e..dc61b125f 100644
--- a/tools/rosbag/src/rosbag/bag.py
+++ b/tools/rosbag/src/rosbag/bag.py
@@ -50,8 +50,8 @@ import threading
 import time
 import yaml
 
-from Cryptodome.Cipher import AES
-from Cryptodome.Random import get_random_bytes
+from Crypto.Cipher import AES
+from Crypto.Random import get_random_bytes
 
 import gnupg
```
I can submit a PR but I am not sure how specific to gentoo this is.

jinmenglei, 2021-02-20:
How about this

```
try:
    from Cryptodome.Cipher import AES
except ImportError:
    rospy.logwarn(
        'Failed to load Python extension for Cryptodome.Cipher support. '
        'Crypto.Cipher will be available.')
    from Crypto.Cipher import AES
```

JiaoxianDu, 2024-08-15:
pip install pycryptodomex

closed 2025-05-31 by sloretz