ros/ros_comm #2146: use rosnode_ping occur exception: BadStatusLine

HomeAventador, 2021-03-26:
File "/home/mdc/git/ads1_0_test/autotest/lib/SoftwareQT_lib/kybersim_lib/tools/node_checker.py", line 119, in check_node_cleanup
    if rosnode.rosnode_ping(node, max_count=1):
  File "/opt/x86/ros/kinetic/lib/python2.7/dist-packages/rosnode/__init__.py", line 326, in rosnode_ping
    pid = _succeed(node.getPid(ID))
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1313, in single_request
    response = h.getresponse(buffering=True)
  File "/usr/lib/python2.7/httplib.py", line 1182, in getresponse
    response.begin()
  File "/usr/lib/python2.7/httplib.py", line 467, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python2.7/httplib.py", line 431, in _read_status
    raise BadStatusLine(line)
BadStatusLine: ''

fujitatomoya, 2021-03-28:
thanks for opening issue but we would need more information like precise reproducible procedure.

i did check that rosnode ping works okay with following environment.

- noetic with https://github.com/ros/ros_comm/commit/bf56b4022d836c8ce63e0abf2474ade32f9a8f64
- talker node available with `rosnode ping /talker`
- talker node available with `rosnode ping -a`

so far, i do not see any problems for that.

jacobperron, 2021-04-06:
As @fujitatomoya mentioned, without steps to reproduce the issue it is difficult to troubleshoot.

@HomeAventador Can you provide a simple set of steps that we can use to reproduce?

HomeAventador, 2021-04-21:
First, init node A. Then start some nodes in A.
There're about 20 nodes in environment, other nodes (exclude A) repeated restart. Each startup complete, i try to excute "rosnode ping -a" in node A.
The probability of this problem is 1 in 300.

HomeAventador, 2021-04-21:
Problem occur about 20 times. Each raise exception around rosnode ping A.

closed 2025-05-31 by sloretz