ros/ros_comm #2153: Missing rospy API documentation

brianhou, 2021-04-07:
Hi there,

I'm trying to find the rospy API docs, which seem to be missing from http://docs.ros.org/en/api/rospy/html/index.html. Is that the right link, or is there another place to read them?

(Just to catalog my steps to that broken page: I started at https://wiki.ros.org/rospy and followed the link under API Reference to http://docs.ros.org/en/api/rospy/html/.)

Thank you!

VismayVakharia, 2021-04-08:
This works - [http://docs.ros.org/en/melodic/api/rospy/html/rospy-module.html](http://docs.ros.org/en/melodic/api/rospy/html/rospy-module.html)

brianhou, 2021-04-08:
Thank you! That's really helpful, although I was hoping to get documentation for Noetic (and the corresponding link doesn't seem to be available).

fcolecumberri, 2021-04-30:
is there any tutorial about how to generate the rospy documentation? even if I need to make a local instance, that would be useful

gavanderhoorn, 2021-06-12:
Is this https://github.com/ros-infrastructure/rosdoc_lite/issues/98?

closed 2025-05-31 by sloretz