ros/ros_comm #2176: ResourceWarning with rosnode ping

git-afsantos, 2021-08-17:
Greetings!

I have some test code in Python using `unittest` and `rosnode.ping_node()`.
When running this test code with Python 3 (ROS Noetic), a series of warnings are displayed, stating that sockets are not explicitly closed. I have not seen these warnings before with ROS Melodic and Python 2.7, so I am guessing this has been added in Python 3.x.

For an example, consider the following environment:

```bash
#!/usr/bin/env bash
roscore &
rostopic echo /topic
```

And the following test code:

```python
import sys
import unittest
from rosnode import rosnode_ping_all
import rospy

class TestTracer(unittest.TestCase):
    def test_ping_warning(self):
        rosnode_ping_all(skip_cache=True)

def main():
    rospy.init_node("ping_tester", log_level=rospy.DEBUG)
    unittest.main(module=__name__, argv=[__name__], exit=False)
    sys.stdout.flush()

if __name__ == "__main__":
    main()
```

Executing this code with Python 3 produces the following output.

<details>
  <summary>Test Output</summary>

> [DEBUG] [1629206461.806698]: init_node, name[/ping_tester], pid[46274]
> [DEBUG] [1629206461.809490]: binding to 0.0.0.0 0
> [DEBUG] [1629206461.811671]: bound to 0.0.0.0 33287
> [DEBUG] [1629206461.814433]: ... service URL is rosrpc://andre-ros-noetic:33287
> [DEBUG] [1629206461.816574]: [/ping_tester/get_loggers]: new Service instance
> [DEBUG] [1629206461.820822]: ... service URL is rosrpc://andre-ros-noetic:33287
> [DEBUG] [1629206461.823336]: [/ping_tester/set_logger_level]: new Service instance
> /opt/ros/noetic/lib/python3/dist-packages/rosnode/__init__.py:403: ResourceWarning: unclosed <socket.socket fd=10, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6, laddr=('127.0.0.1', 47764), raddr=('127.0.0.1', 11311)>
>   if rosnode_ping(node, max_count=1, verbose=verbose, skip_cache=skip_cache):
> ResourceWarning: Enable tracemalloc to get the object allocation traceback
> /opt/ros/noetic/lib/python3/dist-packages/rosnode/__init__.py:403: ResourceWarning: unclosed <socket.socket fd=11, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6, laddr=('127.0.0.1', 49064), raddr=('127.0.1.1', 42105)>
>   if rosnode_ping(node, max_count=1, verbose=verbose, skip_cache=skip_cache):
> ResourceWarning: Enable tracemalloc to get the object allocation traceback
> /opt/ros/noetic/lib/python3/dist-packages/rosnode/__init__.py:403: ResourceWarning: unclosed <socket.socket fd=10, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6, laddr=('127.0.0.1', 47768), raddr=('127.0.0.1', 11311)>
>   if rosnode_ping(node, max_count=1, verbose=verbose, skip_cache=skip_cache):
> ResourceWarning: Enable tracemalloc to get the object allocation traceback
> /opt/ros/noetic/lib/python3/dist-packages/rosnode/__init__.py:403: ResourceWarning: unclosed <socket.socket fd=11, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6, laddr=('127.0.0.1', 56510), raddr=('127.0.1.1', 42669)>
>   if rosnode_ping(node, max_count=1, verbose=verbose, skip_cache=skip_cache):
> ResourceWarning: Enable tracemalloc to get the object allocation traceback
> /opt/ros/noetic/lib/python3/dist-packages/rosnode/__init__.py:403: ResourceWarning: unclosed <socket.socket fd=10, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6, laddr=('127.0.0.1', 47772), raddr=('127.0.0.1', 11311)>
>   if rosnode_ping(node, max_count=1, verbose=verbose, skip_cache=skip_cache):
> ResourceWarning: Enable tracemalloc to get the object allocation traceback
> /opt/ros/noetic/lib/python3/dist-packages/rosnode/__init__.py:403: ResourceWarning: unclosed <socket.socket fd=11, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6, laddr=('127.0.0.1', 56428), raddr=('127.0.1.1', 42983)>
>   if rosnode_ping(node, max_count=1, verbose=verbose, skip_cache=skip_cache):
> ResourceWarning: Enable tracemalloc to get the object allocation traceback
> ping_test.py:11: ResourceWarning: unclosed <socket.socket fd=9, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6, laddr=('127.0.0.1', 47762), raddr=('127.0.0.1', 11311)>
>   rosnode_ping_all(skip_cache=True)
> ResourceWarning: Enable tracemalloc to get the object allocation traceback
> .
>
> Ran 1 test in 0.018s
> 
> OK
</details>

Here is an abbreviated version of the issue:

>
>     ResourceWarning: Enable tracemalloc to get the object allocation traceback
>     ping_test.py:11: ResourceWarning: unclosed <socket.socket fd=9, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6, laddr=('127.0.0.1', 47762), raddr=('127.0.0.1', 11311)>
>       rosnode_ping_all(skip_cache=True)
>     ResourceWarning: Enable tracemalloc to get the object allocation traceback

---

The issue can be resolved by enveloping `ServerProxy` instances used in `ping_node` ([example 1](https://github.com/ros/ros_comm/blob/c7c5bcb22780e7f6dbbff9bc923afca34fd612b6/tools/rosnode/src/rosnode/__init__.py#L323), [example 2](https://github.com/ros/ros_comm/blob/c7c5bcb22780e7f6dbbff9bc923afca34fd612b6/tools/rosnode/src/rosnode/__init__.py#L362)) with a context manager, [as stated in the docs](https://docs.python.org/3/library/xmlrpc.client.html#serverproxy-objects) and as shown in similar reports in [StackOverflow](https://stackoverflow.com/q/48160728).

RicardoM17, 2024-12-02:
Hi @git-afsantos Did you end up solving this in some way?

I'm facing the same issue now (this is actually causing a memory leak in my node) so if you have any findings that you could share that would be great.

git-afsantos, 2024-12-12:
Hey, @RicardoM17 .
It has been a long time, so I may not recall correctly, but I think that my final solution just ended up dropping `ping` entirely.

closed 2025-04-29 by sloretz