ros/ros_comm #2182: regression from #2132: ros::param::set limited to 32 kB on localhost on computers with kernel 4.x

peci1, 2021-09-01:
Minimum non-working example:

```C++

int main(int argc, char **argv)
{
  ros::init(argc, argv, "node");
  ros::param::set("param", std::string(32654, 'a'));
  return 0;
}

```

Running this program with a ROS master on localhost results in the following error message on some computers:

    [ERROR 1630469544.713298952 /node] [setParam] Failed to contact master at [localhost:11311].  Retrying...

It seems that computers running a 4.x kernel have this 32 kB limit whereas 5.x kernels allow 100s of MBs.

Affected:
- Kontron KTQM87 with Core i7 4th gen (amd64 platform, Intel CPU, kernel 4.15)
- NVidia Xavier AGX (arm64 platform, Maxwell CPU, kernel 4.9)

Non-affected:
- Intel NUC 10th gen (amd64 platform, Intel CPU, kernel 5.4)
- Lenovo T14s notebook (amd64 platform, AMD Ryzen CPU, kernel 5.13)

`ulimit -a` doesn't show any related limitation:

```
core file size          (blocks, -c) unlimited
data seg size           (kbytes, -d) unlimited
scheduling priority             (-e) 40
file size               (blocks, -f) unlimited
pending signals                 (-i) 63543
max locked memory       (kbytes, -l) 65536
max memory size         (kbytes, -m) unlimited
open files                      (-n) 1024
pipe size            (512 bytes, -p) 8
POSIX message queues     (bytes, -q) 819200
real-time priority              (-r) 99
stack size              (kbytes, -s) 8192
cpu time               (seconds, -t) unlimited
max user processes              (-u) 63543
virtual memory          (kbytes, -v) unlimited
file locks                      (-x) unlimited
```

Decreasing the payload size to 32653 or less makes the program work on all computers.

I tried running the program with XmlRpc client debug logs and here's what I get around the `setParam` call:

```
XmlRpcClient::execute: method setParam (_connectionState IDLE).
XmlRpcClient::generateRequest: header is 115 bytes, content-length is 32654.
XmlRpcClient::writeRequest (attempt 1):
POST / HTTP/1.1
User-Agent: XMLRPC++ 0.7
Host: localhost:11311
Content-Type: text/xml
Content-length: 32654

<?xml version="1.0"?>
<methodCall><methodName>setParam</methodName>
<params><param><value>/node</value></param><param><value>/node/param</value></param><param><value>aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
XmlRpcSocket::nbWrite: send/write returned 32768.
XmlRpcSocket::nbWrite: send/write returned -1.
XmlRpcClient::close: fd 3.
XmlRpcSource::close: closing socket 3.
XmlRpcSocket::close: fd 3.
XmlRpcSource::close: done closing socket 3.
[ERROR 1630465298.375945115 /node] [setParam] Failed to contact master at [localhost:11311].  Retrying...
```

The two `nbWrite` are interesting - the second one ends with error, but the error message is not outputted. I tried manually reading errno in this case and the error is `EPIPE`. According to the docs that means that the other side (rosmaster) has closed the connection before it read all the data.

That would point towards a bug in rosmaster, but if I run a similar program using rospy, then I can set parameters as large as I want. So it is probably some incompatibility between the C++ XMLRPC client and Python XMLRPC server.

Also, if ROS master is running on a different computer in LAN, no combination of master/client computers leads to this error. It is solely affecting the case when master and the node are running on the same machine.

Last, if I edit the patch from https://github.com/ros/ros_comm/pull/2132 and force HTTP/1.1 even on the 4.x kernels, the 32 kB limit is away. @jikawa-az @fujitatomoya @emersonknapp any ideas how to get both the fix from #2132 and allow larger params again?

emersonknapp, 2021-09-01:
Out of curiosity, were you using this functionality on Kinetic, or is that older than your application? #2132 was fixing a serious performance problem introduced to Lunar by https://github.com/ros/ros_comm/pull/1287 - but it seems that this is a complicated issue with many facets. 

Are there other types of communications that have new size problems, such as messages or service calls? Or is it just parameters?

emersonknapp, 2021-10-20:
I was not able to quantify it, only reliably demonstrate rosmaster locking up in that toy stress app - the behavior we observed in a more complex customer application. This is a tricky one!

jacobperron, 2021-11-05:
@wjwwood @clalancette thoughts?

fujitatomoya, 2021-11-05:
@peci1 sorry to be late to respond this, could you try the following fix with your environment? since i do not have reproducible environment with kernel version under 4.15.

```diff
diff --git a/utilities/xmlrpcpp/src/XmlRpcClient.cpp b/utilities/xmlrpcpp/src/XmlRpcClient.cpp
index b85424144..936ce01fa 100644
--- a/utilities/xmlrpcpp/src/XmlRpcClient.cpp
+++ b/utilities/xmlrpcpp/src/XmlRpcClient.cpp
@@ -328,7 +328,7 @@ std::string
 XmlRpcClient::generateHeader(size_t length) const
 {
   std::string header =
-    "POST " + _uri + " HTTP/1.1\r\n"
+    "POST " + _uri + " HTTP/1.0\r\n"
     "User-Agent: ";
   header += XMLRPC_VERSION;
   header += "\r\nHost: ";
```

jokla, 2021-11-08:
Hi,

I have the same problem on the NVidia Xavier AGX (4.9.201-tegra) with [apriltag_ros](https://github.com/AprilRobotics/apriltag_ros/) when loading a custom tags file https://github.com/AprilRobotics/apriltag_ros/blob/master/apriltag_ros/config/tags.yaml.

Thanks for looking for a fix!

fujitatomoya, 2021-11-08:
@jokla did you confirm my suggested fix resolve the problem?

jokla, 2021-11-08:
Hi @fujitatomoya. I can test it in the next few days, hopefully tomorrow.

jokla, 2021-11-09:
Hi @fujitatomoya, I tested the fix on the Xavier but I still have the same issue.
This is what I did: I cloned `ros_comm` into a catkin workspace, changed the file `XmlRpcClient.cpp` and build it. I sourced my workspace, I checked with ldd the dependencies of the `apriltag_ros` application and it is using libxmlrpcpp.so from my workspace.

peci1, 2021-11-09:
This is weird. I'm no longer able to repro this issue on any of our computers. I've event started up the original Jetson on which I discovered this bug. The Jetson has not been updated since then (I'm almost sure with that). The minimum non-working example from the initial post is working without issues even without any bugfix :/

peci1, 2021-11-09:
@jokla would you be able to post here the non-working example you're running?

jokla, 2021-11-10:
@peci1 How many times did you try to run the example? For me, apriltag_ros works 7/10 times with and without @fujitatomoya fix.
Just to make sure I am doing the correct thing, these are the steps:
1) Clone `ros_comm` on my workspace, checkout `melodic-devel` branch, apply @fujitatomoya fix and build it
2) Source from my workspace install folder
3) Run apriltag_ros with a custom yaml file

I checked with ldd the apriltag_ros executable dependencies and it is using `libxmlrpcpp.so` from my workspace.

jokla, 2021-11-10:
It looks like the problem is only occurring when I use a nodelet.

Non working example:
1) install apriltag with `sudo apt-get install ros-melodic-apriltag-ros`
2) Download [apriltag_nodelet.launch](https://gist.github.com/jokla/4f54be23cc7379afe2b86f58fb42eb11/raw/4a166e364f86748252dbb81da1f560000796accc/apriltag_nodelet.launch) and [tags.yaml ](https://gist.github.com/jokla/4f54be23cc7379afe2b86f58fb42eb11/raw/4a166e364f86748252dbb81da1f560000796accc/tags.yaml) 
3) Open apriltag_nodelet.launch and modify the path to the tags.yaml file 
`<rosparam command="load" file="/home/nvidia/Desktop/tags.yaml"/>`
4) Run the application with `roslaunch apriltag_nodelet.launch`

This is what I get:
```
auto-starting new master
process[master]: started with pid [16834]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 72170d60-421a-11ec-8342-48b02d4d6183
process[rosout-1]: started with pid [16877]
started core service [/rosout]
process[camera_nodelet_manager-2]: started with pid [16881]
process[apriltag_ros-3]: started with pid [16889]
[/apriltag_ros] [ INFO] [1636544191.115069223]: Loading nodelet /apriltag_ros of type apriltag_ros/ContinuousDetector to manager camera_nodelet_manager with the following remappings:
[/apriltag_ros] [ERROR] [1636544191.127288945]: [setParam] Failed to contact master at [localhost:11311].  Retrying...
[/camera_nodelet_manager] [ INFO] [1636544191.146133512]: Initializing nodelet with 8 worker threads.
```

peci1, 2021-11-10:
Great, @jokla, your example can trigger the bug on my Jetson, too!

When running as a standalone nodelet, I could not trigger it. So it only happens when loaded as a nodelet into a nodelet manager.

peci1, 2021-11-10:
I can also trigger it even with the nodelet manager already running and only loading the nodelet into it.

peci1, 2021-11-10:
I've updated the minimum non-working example in the first comment on this issue. @jokla , can you verify it also fails for you? Only this time, the parameter size limit is not 32kB, but 22083 bytes.

You can build and run the example with:

    g++ -I/opt/ros/${ROS_DISTRO}/include -c rosparam.c
    g++ rosparam.o /opt/ros/${ROS_DISTRO}/lib/libroscpp.so
    timeout 1 ./a.out

It either writes nothing (success), or prints the setParam error (fail).

jokla, 2021-11-10:
@peci1  I confirm that the new example is also failing for me.

peci1, 2021-11-10:
@fujitatomoya I can also confirm the patch you suggested does not help.

peci1, 2021-11-10:
@fujitatomoya The patch you suggested actually brings the problem back even when using HTTP 1.1 in rosgraph/xmlrpc.py .

Info from debugging rosmaster with the minimum non-working example and HTTP 1.0 in rosgraph/xmlrpc.py (and without the xmlrpcpp patch):

1. When the `setParam()` request fails, the rosmaster does never dispatch [`ROSMasterHandler::setParam()`](https://github.com/ros/ros_comm/blob/9bec9d113196a4389d1dfa574dee882172823f71/tools/rosmaster/src/rosmaster/master_api.py#L356).

2. It never succeeds on a fresh rosmaster. The second run on the same master may succeed.

3. strace of rosmaster child threads spawned when calling the test script shows that after `hasParam("/use_sime_time")`, there is no other `select()` call waking up. Usually, in the non-failed cases, the failing `setParam()` follows right after this `hasParam()`. So it seems that on the rosmaster side, there is nothing to do wrong, because the very deepest `select()` call is hiding the incoming request from the rosmaster.

4. It seems that the major difference between HTTP 1.0 and 1.1 is connection reuse (see the straces)

cleaned up strace of the failing case:

<details>

```
[pid  5940] clone(strace: Process 5944 attached
 <unfinished ...>
[pid  5944] set_robust_list(0x7f67fff250, 24 <unfinished ...>
[pid  5944] <... set_robust_list resumed> ) = 0
[pid  5944] mmap(0x7f64000000, 67108864, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_NORESERVE, -1, 0) = 0x7f5c000000
[pid  5944] mprotect(0x7f5c000000, 135168, PROT_READ|PROT_WRITE) = 0
[pid  5944] futex(0x7f600027f0, FUTEX_WAKE_PRIVATE, 1) = 1
[pid  5944] setsockopt(5, SOL_TCP, TCP_NODELAY, [1], 4 <unfinished ...>
[pid  5944] <... setsockopt resumed> )  = 0
[pid  5944] futex(0x55b1429560, FUTEX_WAKE_PRIVATE, 1) = 1
[pid  5944] futex(0x55b1429560, FUTEX_WAKE_PRIVATE, 1 <unfinished ...>
[pid  5944] <... futex resumed> )       = 0
[pid  5944] recvfrom(5, "POST / HTTP/1.0\r\nUser-Agent: XMLRPC++ 0.7\r\nHost: localhost:11311\r\nContent-Type: text/xml\r\nContent-length: 181\r\n\r\n<?xml version=\"1.0\"?>\r\n<methodCall><methodName>hasParam</methodName>\r\n<params><param><value>/node</value></param><param><value>/tcp_keepalive</value></param></params></methodCall>\r\n", 8192, 0, NULL, NULL) = 294
[pid  5944] getrandom("\xac\xb4\x58\x54\xa5\xf9\xf5\xe5", 8, GRND_NONBLOCK) = 8
[pid  5944] openat(AT_FDCWD, "/usr/lib/python2.7/encodings/ascii.py", O_RDONLY) = 6
[pid  5944] fstat(6, {st_mode=S_IFREG|0644, st_size=1248, ...}) = 0
[pid  5944] openat(AT_FDCWD, "/usr/lib/python2.7/encodings/ascii.pyc", O_RDONLY) = 7
[pid  5944] fstat(7, {st_mode=S_IFREG|0644, st_size=2261, ...}) = 0
[pid  5944] read(7, "\3\363\r\n\2a:`c\0\0\0\0\0\0\0\0\3\0\0\0@\0\0\0s\273\0\0\0d\0\0Z\0\0d\1\0d\2\0l\1\0Z\1\0d\3\0e\1\0j\2\0f\1\0d\4\0\204\0\0\203\0\0YZ\2\0d\5\0e\1\0j\3\0f\1\0d\6\0\204\0\0\203\0\0YZ\3\0d\7\0e\1\0j\4\0f\1\0d\10\0\204\0\0\203\0\0YZ\4\0d\t\0e\2\0e\1\0j\5\0f\2\0d\n\0\204\0\0\203\0\0YZ\5\0d\v\0e\2\0e\1\0j\6\0f\2\0d\f\0\204\0\0\203\0\0YZ\6\0d\r\0e\5\0e\6\0f\2\0d\16\0\204\0\0\203\0\0YZ\7\0d\17\0\204\0\0Z\10\0d\2\0S(\20\0\0\0s\201\0\0\0 Python 'ascii' Codec\n\n\nWritten by Marc-Andre Lemburg (mal@lemburg.com).\n\n(c) Copyright CNRI,"..., 4096) = 2261
[pid  5944] fstat(7, {st_mode=S_IFREG|0644, st_size=2261, ...}) = 0
[pid  5944] read(7, "", 4096)           = 0
[pid  5944] close(7)                    = 0
[pid  5944] close(6)                    = 0
[pid  5944] sendto(5, "HTTP/1.0 200 OK\r\nServer: BaseHTTP/0.3 Python/2.7.17\r\nDate: Wed, 10 Nov 2021 15:47:06 GMT\r\nContent-type: text/xml\r\nContent-length: 249\r\n\r\n<?xml version='1.0'?>\n<methodResponse>\n<params>\n<param>\n<value><array><data>\n<value><int>1</int></value>\n<value><string>/tcp_keepalive</string></value>\n<value><boolean>0</boolean></va"..., 386, 0, NULL, 0) = 386
[pid  5944] shutdown(5, SHUT_WR)        = 0
[pid  5944] close(5)                    = 0
[pid  5944] madvise(0x7f677ff000, 8253440, MADV_DONTNEED) = 0
[pid  5944] exit(0)                     = ?
[pid  5944] +++ exited with 0 +++
[pid  5940] clone(strace: Process 5949 attached
 <unfinished ...>
[pid  5949] set_robust_list(0x7f67fff250, 24 <unfinished ...>
[pid  5949] <... set_robust_list resumed> ) = 0
[pid  5949] futex(0x7f600025a0, FUTEX_WAKE_PRIVATE, 1 <unfinished ...>
[pid  5949] <... futex resumed> )       = 0
[pid  5949] futex(0x55b1429560, FUTEX_WAKE_PRIVATE, 1 <unfinished ...>
[pid  5949] <... futex resumed> )       = 0
[pid  5949] futex(0x7f600027f0, FUTEX_WAKE_PRIVATE, 1) = 0
[pid  5949] setsockopt(5, SOL_TCP, TCP_NODELAY, [1], 4 <unfinished ...>
[pid  5949] <... setsockopt resumed> )  = 0
[pid  5949] futex(0x55b1429560, FUTEX_WAKE_PRIVATE, 1) = 1
[pid  5949] futex(0x55b1429560, FUTEX_WAKE_PRIVATE, 1 <unfinished ...>
[pid  5949] <... futex resumed> )       = 0
[pid  5949] recvfrom(5, "POST / HTTP/1.0\r\nUser-Agent: XMLRPC++ 0.7\r\nHost: localhost:11311\r\nContent-Type: text/xml\r\nContent-length: 292\r\n\r\n<?xml version=\"1.0\"?>\r\n<methodCall><methodName>registerPublisher</methodName>\r\n<params><param><value>/node</value></param><param><value>/rosout</value></param><param><value>rosgraph_msgs/Log</value></param><"..., 8192, 0, NULL, NULL) = 405
[pid  5949] getrandom("\x64\x66\xa5\x78\x12\x24\xf7\x79", 8, GRND_NONBLOCK) = 8
[pid  5949] getpid()                    = 5893
[pid  5949] newfstatat(AT_FDCWD, "/etc/localtime", {st_mode=S_IFREG|0644, st_size=2338, ...}, 0) = 0
[pid  5949] fstat(3, {st_mode=S_IFREG|0664, st_size=50073, ...}) = 0
[... cut ...] <array><data>\n<value><int>1</int></value>\n<value><string>Unregistered [/node] as provider of [/node/set_logger_level]</s"..., 424, 0, NULL, 0) = 424
[pid  7833] recvfrom(5,  <unfinished ...>
[pid  7833] <... recvfrom resumed> "", 8192, 0, NULL, NULL) = 0
[pid  7833] shutdown(5, SHUT_WR)        = 0
[pid  7833] close(5)                    = 0
[pid  7833] madvise(0x7fadffc000, 8253440, MADV_DONTNEED) = 0
[pid  7833] exit(0)                     = ?
[pid  7833] +++ exited with 0 +++
```

</details>

fujitatomoya, 2021-11-13:
@peci1 @jokla thanks, this is deep.... :thinking: 

I am not pretty much sure about the root cause of this, but it seems to be version mismatch.

> The patch you suggested actually brings the problem back even when using HTTP 1.1 in rosgraph/xmlrpc.py

i guess this is because of mismatch between `XmlRpcClient` and `XmlRpcServerConnection`?

How about the following patch? what do you think?

```diff
diff --git a/utilities/xmlrpcpp/src/XmlRpcClient.cpp b/utilities/xmlrpcpp/src/XmlRpcClient.cpp
index b85424144..936ce01fa 100644
--- a/utilities/xmlrpcpp/src/XmlRpcClient.cpp
+++ b/utilities/xmlrpcpp/src/XmlRpcClient.cpp
@@ -328,7 +328,7 @@ std::string
 XmlRpcClient::generateHeader(size_t length) const
 {
   std::string header = 
-    "POST " + _uri + " HTTP/1.1\r\n"
+    "POST " + _uri + " HTTP/1.0\r\n"
     "User-Agent: ";
   header += XMLRPC_VERSION;
   header += "\r\nHost: ";
diff --git a/utilities/xmlrpcpp/src/XmlRpcServerConnection.cpp b/utilities/xmlrpcpp/src/XmlRpcServerConnection.cpp
index da4671981..a4231984b 100644
--- a/utilities/xmlrpcpp/src/XmlRpcServerConnection.cpp
+++ b/utilities/xmlrpcpp/src/XmlRpcServerConnection.cpp
@@ -362,7 +362,7 @@ std::string
 XmlRpcServerConnection::generateHeader(std::string const& body)
 {
   std::string header = 
-    "HTTP/1.1 200 OK\r\n"
+    "HTTP/1.0 200 OK\r\n"
     "Server: ";
   header += XMLRPC_VERSION;
   header += "\r\n"
```

peci1, 2021-11-19:
I've tried it and the behavior is the same :(

fujitatomoya, 2021-11-19:
@peci1 thanks for checking, so this is not really about version mismatch...

@Barry-Xu-2018 @iuhilnehc-ynos do you guys happen to have any clue?

Barry-Xu-2018, 2021-11-22:
@fujitatomoya 

> do you guys happen to have any clue?  

By now we don't find any clue.

iuhilnehc-ynos, 2021-11-24:
@fujitatomoya @Barry-Xu-2018 

I don't know why there is no `close()` syscall on both `parameters` and `rosmaster` in the `strace` log when the socket is lost(that leads to SIGPIPE), but what I care about is whether the `setparam` can retry. Unfortunately, it can not.

Deadlock happened(lock operations twice[[1](https://github.com/ros/ros_comm/blob/9bec9d113196a4389d1dfa574dee882172823f71/clients/roscpp/src/libros/param.cpp#L79),[2](https://github.com/ros/ros_comm/blob/9bec9d113196a4389d1dfa574dee882172823f71/clients/roscpp/src/libros/param.cpp#L264)] for `g_params_mutex` in a thread) as follows,
<details><summary> callstack </summary>
<p>

```cpp
(gdb) bt
#0  __lll_lock_wait () at ../sysdeps/unix/sysv/linux/x86_64/lowlevellock.S:135
#1  0x00007f2ea4f3f025 in __GI___pthread_mutex_lock (mutex=0x7f2ea7281260 <ros::param::g_params_mutex>) at ../nptl/pthread_mutex_lock.c:80
#2  0x00007f2ea6e9f222 in boost::posix::pthread_mutex_lock (m=0x7f2ea7281260 <ros::param::g_params_mutex>) at /usr/include/boost/thread/pthread/mutex.hpp:62
#3  boost::mutex::lock (this=0x7f2ea7281260 <ros::param::g_params_mutex>) at /usr/include/boost/thread/pthread/mutex.hpp:116
#4  0x00007f2ea6ea0379 in boost::unique_lock<boost::mutex>::lock (this=0x7ffc2f33bac0) at /usr/include/boost/thread/lock_types.hpp:346
#5  0x00007f2ea6ea02a7 in boost::unique_lock<boost::mutex>::unique_lock (this=0x7ffc2f33bac0, m_=...) at /usr/include/boost/thread/lock_types.hpp:124
#6  0x00007f2ea6ebf4a6 in ros::param::getImpl (key="/rosout_disable_topics_generation", v=..., use_cache=true)
    at /root/ros_catkin_ws/src/ros_comm/roscpp/src/libros/param.cpp:264
    
    (second lock)
    
#7  0x00007f2ea6ebfecb in ros::param::getImpl (key="/rosout_disable_topics_generation", b=@0x55841219e6d1: false, use_cache=true)
    at /root/ros_catkin_ws/src/ros_comm/roscpp/src/libros/param.cpp:401
#8  0x00007f2ea6ec0123 in ros::param::getCached (key="/rosout_disable_topics_generation", b=@0x55841219e6d1: false)
    at /root/ros_catkin_ws/src/ros_comm/roscpp/src/libros/param.cpp:461
#9  0x00007f2ea6f745df in ros::ROSOutAppender::log (this=0x55841219e610, level=ros::console::levels::Error, 
    str=0x5584121920f0 "[setParam] Failed to contact master at [172.104.82.232:11311].  Retrying...", 
    file=0x7f2ea6fb2848 "/root/ros_catkin_ws/src/ros_comm/roscpp/src/libros/master.cpp", function=0x558412195d40 "master::execute", line=202)
    at /root/ros_catkin_ws/src/ros_comm/roscpp/src/libros/rosout_appender.cpp:111
#10 0x00007f2ea4771833 in ros::console::impl::Log4cxxAppender::append (this=0x55841219ea10, event=..., pool=...)
    at /root/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:347
#11 0x00007f2ea4257665 in log4cxx::AppenderSkeleton::doAppend(log4cxx::helpers::ObjectPtrT<log4cxx::spi::LoggingEvent> const&, log4cxx::helpers::Pool&) ()
   from /usr/lib/x86_64-linux-gnu/liblog4cxx.so.10
#12 0x00007f2ea425560f in log4cxx::helpers::AppenderAttachableImpl::appendLoopOnAppenders(log4cxx::helpers::ObjectPtrT<log4cxx::spi::LoggingEvent> const&, log4cxx::helpers::Pool&) () from /usr/lib/x86_64-linux-gnu/liblog4cxx.so.10
#13 0x00007f2ea429aadd in log4cxx::Logger::callAppenders(log4cxx::helpers::ObjectPtrT<log4cxx::spi::LoggingEvent> const&, log4cxx::helpers::Pool&) const ()
   from /usr/lib/x86_64-linux-gnu/liblog4cxx.so.10
#14 0x00007f2ea429ad0e in log4cxx::Logger::forcedLog(log4cxx::helpers::ObjectPtrT<log4cxx::Level> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, log4cxx::spi::LocationInfo const&) const () from /usr/lib/x86_64-linux-gnu/liblog4cxx.so.10
#15 0x00007f2ea476e935 in ros::console::impl::print (handle=0x558412194010, level=ros::console::levels::Error, 
    str=0x558412174840 "[setParam] Failed to contact master at [172.104.82.232:11311].  Retrying...", 
    file=0x7f2ea6fb2848 "/root/ros_catkin_ws/src/ros_comm/roscpp/src/libros/master.cpp", 
    function=0x7f2ea6fb2b20 <ros::master::execute(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, XmlRpc::XmlRpcValue const&, XmlRpc::XmlRpcValue&, XmlRpc::XmlRpcValue&, bool)::__PRETTY_FUNCTION__> "bool ros::master::execute(const string&, const XmlRpc::XmlRpcValue&, XmlRpc::XmlRpcValue&, XmlRpc::XmlRpcValue&, bool)", line=202) at /root/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:187
#16 0x00007f2ea605c487 in ros::console::print (filter=0x0, logger_handle=0x558412194010, level=ros::console::levels::Error, 
    file=0x7f2ea6fb2848 "/root/ros_catkin_ws/src/ros_comm/roscpp/src/libros/master.cpp", line=202, 
    function=0x7f2ea6fb2b20 <ros::master::execute(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, XmlRpc::XmlRpcValue const&, XmlRpc::XmlRpcValue&, XmlRpc::XmlRpcValue&, bool)::__PRETTY_FUNCTION__> "bool ros::master::execute(const string&, const XmlRpc::XmlRpcValue&, XmlRpc::XmlRpcValue&, XmlRpc::XmlRpcValue&, bool)", fmt=0x7f2ea6fb2950 "[%s] Failed to contact master at [%s:%d].  %s") at /root/ros_catkin_ws/src/rosconsole/src/rosconsole/rosconsole.cpp:552
#17 0x00007f2ea6e90225 in ros::master::execute (method="setParam", request=..., response=..., payload=..., wait_for_master=true)
    at /root/ros_catkin_ws/src/ros_comm/roscpp/src/libros/master.cpp:202
#18 0x00007f2ea6ebe87a in ros::param::set (key="param_0_1_1_1", v=...) at /root/ros_catkin_ws/src/ros_comm/roscpp/src/libros/param.cpp:81

    (first lock)

[... cut ...]
   if (g_subscribed_params.find(clean_key) != g_subscribed_params.end())
   {
@@ -813,7 +813,7 @@ void unsubscribeCachedParam(const std::string& key)
 void unsubscribeCachedParam(void)
 {
   // lock required, all of the cached parameter will be unsubscribed.
-  boost::mutex::scoped_lock lock(g_params_mutex);
+  boost::recursive_mutex::scoped_lock lock(g_params_mutex);
 
   for(S_string::iterator itr = g_subscribed_params.begin();
     itr != g_subscribed_params.end(); ++itr)
```

</p>
</details>

iuhilnehc-ynos, 2021-11-24:
After using the patch, I tested the application to set param with `nh.setParam("param_final", std::string("b") + std::string(32769000, 'a') + std::string("e"));` on `Linux localhost 4.15.0-156-generic #163-Ubuntu SMP Thu Aug 19 23:31:58 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`, it can work as expected (`rosparam get /param_final > param.txt 2>&1`).

peci1, 2021-11-24:
@iuhilnehc-ynos What you refer to is actually reported bug #2181 . However, getting rid of this bug with your patch lets the parameter get set in the end. It's probably because of some kind of retry, because the first attempt to set the parameter fails, then the `cannot communicate with master` error is written out, and then there is some kind of retry that succeeds. So there is still the problem that the first try fails and the error is written to the terminal - but practically, solving the rosparam deadlock gets the parameter set. I still think there should be a more proper solution, but I think getting the suggested patch merged will at least solve the problem practically.

Here's the XmlRpcClient log with the patch applied:

<details>

```
XmlRpcClient::execute: method setParam (_connectionState IDLE).
XmlRpcClient::generateRequest: header is 116 bytes, content-length is 230203.
XmlRpcClient::writeRequest (attempt 1):
POST / HTTP/1.1
User-Agent: XMLRPC++ 0.7
Host: localhost:11311
Content-Type: text/xml
Content-length: 230203

<?xml version="1.0"?>
<methodCall><methodName>setParam</methodName>
<params><param><value>/node</value></param><param><value>/param</value></param><param><value>aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
XmlRpcSocket::nbWrite: send/write returned 22400, nToWrite = 230319.
XmlRpcSocket::nbWrite: send/write returned -1, nToWrite = 207919.
XmlRpcSocket::nbWrite: send/write errno 32.
XmlRpcClient::close: fd 3.
XmlRpcSource::close: closing socket 3.
XmlRpcSocket::close: fd 3.
XmlRpcSource::close: done closing socket 3.
[ERROR] [1637750469.385086522]: [setParam] Failed to contact master at [localhost:11311].  Retrying...
XmlRpcClient new client: host localhost, port 11311.
XmlRpcClient::execute: method subscribeParam (_connectionState NO_CONNECTION).
XmlRpcClient::doConnect: fd 3.
found host as 127.0.0.1

XmlRpcClient::generateRequest: header is 113 bytes, content-length is 268.
XmlRpcClient::writeRequest (attempt 1):
POST / HTTP/1.1
User-Agent: XMLRPC++ 0.7
Host: localhost:11311
Content-Type: text/xml
Content-length: 268

<?xml version="1.0"?>
<methodCall><methodName>subscribeParam</methodName>
<params><param><value>/node</value></param><param><value>http://clone-robot-jetson:45849/</value></param><param><value>/rosout_disable_topics_generation</value></param></params></methodCall>

XmlRpcSocket::nbWrite: send/write returned 381, nToWrite = 381.
XmlRpcClient::writeRequest: wrote 381 of 381 bytes.
XmlRpcSocket::nbRead: read/recv returned -1.
XmlRpcClient::readHeader: client has read 0 bytes
XmlRpcSocket::nbRead: read/recv returned 429.
XmlRpcSocket::nbRead: read/recv returned 0.
XmlRpcClient::readHeader: client has read 429 bytes
client read content length: 292
XmlRpcClient::readResponse (read 292 bytes)
response:
<?xml version='1.0'?>
<methodResponse>
<params>
<param>
<value><array><data>
<value><int>1</int></value>
<value><string>Subscribed to parameter [/rosout_disable_topics_generation]</string></value>
<value><struct>
</struct></value>
</data></array></value>
</param>
</params>
</methodResponse>

XmlRpcClient::execute: method subscribeParam completed.
XmlRpcClient::execute: method getParam (_connectionState IDLE).
[... 64 lines cut ...]
XmlRpcClient::readHeader: client has read 0 bytes
XmlRpcSocket::nbRead: read/recv returned 384.
XmlRpcSocket::nbRead: read/recv returned -1.
XmlRpcClient::readHeader: client has read 384 bytes
client read content length: 247
XmlRpcClient::readResponse (read 247 bytes)
response:
<?xml version='1.0'?>
<methodResponse>
<params>
<param>
<value><array><data>
<value><int>1</int></value>
<value><string>parameter /param set</string></value>
<value><int>0</int></value>
</data></array></value>
</param>
</params>
</methodResponse>

XmlRpcClient::execute: method setParam completed.
[ INFO] [1637750469.455437532]: Connected to master at [localhost:11311]
```

</details>

peci1, 2021-11-24:
Also, the suggested patch does not break the performance improvement from #2132 (at least on 4.15 kernel).

iuhilnehc-ynos, 2021-11-25:
I'll find a better way (If the response data from `rosmaster` contains 'HTTP1.0', the request socket of `roscpp` must be closed. TODO)  to fix this, but I think that the 'dead lock' should be fixed also.

fujitatomoya, 2021-11-30:
@iuhilnehc-ynos 

recursive deadlock and api negotiation between client and server (this is related to `persistent connection` introduced by HTTP 1.1, right?), both make sense to me. would you consider creating PR against each issue accordingly?

closed 2021-12-03 by jacobperron