ros/ros_comm #2193: ROSCPP_DECL?

juansuerorobotics, 2021-10-18:
in /opt/ros/melodic/include/ros/init.h  
i can see the definition for init

ROSCPP_DECL void init(int &argc, char **argv, const std::string& name, uint32_t options = 0);
 
but when i text search for ROSCPP_DECL under /opt/ros/melodic

all i get is this code in /opt/ros/melodic/include/ros/common.h

// Import/export for windows dll's and visibility for gcc shared libraries.

#ifdef ROS_BUILD_SHARED_LIBS // ros is being built around shared libraries
  #ifdef roscpp_EXPORTS // we are building a shared lib/dll
    #define ROSCPP_DECL ROS_HELPER_EXPORT
  #else // we are using shared lib/dll
    #define ROSCPP_DECL ROS_HELPER_IMPORT
  #endif
#else // ros is being built around static libraries
  #define ROSCPP_DECL
#endif

where is the code that represents ROSCPP_DECL?  is it fed into the compiler on the commandline?
and what kind of code is it?  is it just something like "const auto"  or something ( im newbie c++ )

i see this ROSCPP_DECL on class declarations too like

class ROSCPP_DECL NodeHandle

I searched 
https://github.com/ros/ros_comm/blob/melodic-devel/clients/roscpp/src/libros/init.cpp for ROSCPP_DECL

and it actually dissapears in the CPP file

e.g.
void init(const M_string& remappings, const std::string& name, uint32_t options)

please help me understand thanks.

HaFred, 2021-11-09:
Any updates on this?

juansuerorobotics, 2021-11-09:
Unfortunately no.  Seems like an easy answer except for newbies like me.  Its just a feature of the language i dont understand the coding strategy around.  Oh well.

juansuerorobotics, 2021-11-09:
If i had to guess at this point its something use at compile time to control the visibility of their API.  basically maybe so that you dont use the wrong thing deeper in their code base directly.  So in Bazel or CMAKE or any build system you can set the compile time flags from the command line or from inside the build files.   But thats a wild guess

closed 2025-05-31 by sloretz