ros/ros_comm #2212: AttributeError: 'NoneType' object has no attribute 'close'

MatthijsBurgh, 2021-12-21:
I have no clue of the following exception is caused by incorrect implemetnation on my side or caused by a bug in rospy.
```
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_pubsub.py", line 185, in robust_connect_subscriber
    conn.receive_loop(receive_cb)	    
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 846, in receive_loop
    self.close()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 858, in close
    self.socket.close()
AttributeError: 'NoneType' object has no attribute 'close'
```

MatthijsBurgh, 2022-01-09:
@jacobperron friendly ping

jacobperron, 2022-01-10:
The error message makes it sound like it's a bug in rospy.

@MatthijsBurgh Can you provide simple steps to reproduce the error?

MatthijsBurgh, 2022-01-20:
@jacobperron I will get back to you when I have been able to isolate the issue into a reproducible error.

The situation in which I experienced this error the first time, was in a node launching other nodes.

Now I also experience this in nodes with an action_server client. In both situations it happens during stopping of a node.
```
Exception in thread /hero/local_planner/action_server/result:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_pubsub.py", line 185, in robust_connect_subscriber
    conn.receive_loop(receive_cb)	    
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 846, in receive_loop
    self.close()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 858, in close
    self.socket.close()
AttributeError: 'NoneType' object has no attribute 'close'
```

MichaelGrupp, 2022-02-01:
This is most likely a race condition due to a previous `self.close()` where the socket is set to `None` [here](https://github.com/ros/ros_comm/blob/3f8eca4a1f13996b33c03a5c4ee3a69fb0db3668/clients/rospy/src/rospy/impl/tcpros_base.py#L860). Given the various places where `self.close()` is called because of errors (like maybe improper shutdown in your case) without proper thread sync, I wouldn't be surprised if it crashes sometimes.

madmage, 2022-03-15:
The complete solution of this issue would require to rewrite the classes to be thread safe. Anyway, a opened a PR with a workaround here: https://github.com/ros/ros_comm/pull/2233

MatthijsBurgh, 2022-05-30:
@jacobperron What do you think about #2233? Or would like to go for some locks?

jacobperron, 2022-05-31:
I've merged the workaround. Thanks @madmage!

lyh458, 2022-07-13:
Hi there,
The same error raised when I use a simple function `rospy.wait_for_message`. I called this function in a sub-thread when I met this error first time, I thought it's the multi-thread caused. Then I made a simple test (single nodes without multi-thread), but same error raised. Here is the test code:

```python
import rospy
from sensor_msgs.msg import JointState

import rospy
from sensor_msgs.msg import JointState

def main():
    rospy.init_node('test', anonymous=True)
    while not rospy.is_shutdown():
        js = rospy.wait_for_message('/iiwa/joint_states', JointState, timeout=5)
        print(list(js.position))

if __name__ == '__main__':
    main()
```
As the workaround in #2233 ,  a multi-thread set needed, but how can do it in single node and single thread case?

Thanks in advance!