ros/ros_comm #2216: rospy get_param('') behaves differently after rospy.loginfo()

reinzor, 2022-01-03:
```
$ python -c "import rospy; rospy.init_node('test'); print(rospy.get_param('')); rospy.loginfo(''); print(rospy.get_param(''))"
{'run_id': 'a561b8ca-6c8e-11ec-9979-4737778163b4', 'roslaunch': {'uris': {'host_ws_180018__39993': 'http://ws-180018:39993/'}}, 'rosversion': '1.15.13\n', 'rosdistro': 'noetic\n'}
[INFO][/test][1641212098.262899]: 
{'rosout_disable_topics_generation': {}}
```

Related to [this line](https://github.com/ros/ros_comm/blob/f5fa3a168760d62e9693f10dcb9adfffc6132d22/clients/rospy/src/rospy/impl/rosout.py#L91). When the `get_param_cached` call is removed, the result is the same.

flixr, 2022-03-01:
I'm running in the same issue.
This is a regression caused by #1881.

mikepurvis, 2022-03-01:
@paulbovbel Are you aware?

StephanHasler, 2022-03-03:
Same issue here.
I'm using noetic.

flixr, 2022-03-03:
My current workaround:
```
def ros_get_params():
    """
    This is a workaround since rospy.get_param('/') is currently broken.
    See https://github.com/ros/ros_comm/issues/2216
    """
    def build_nested(name, val, d):
        if name.startswith('/'):
            name = name[1:]
        if '/' not in name:
            d[name] = val
        else:
            head, rest = name.split('/', 1)
            if head not in d:
                d[head] = {}
            build_nested(rest, val, d[head])

    ros_params = {}
    for param in rospy.get_param_names():
        pval = rospy.get_param_cached(param)
        build_nested(param, pval, ros_params)
    return ros_params
```