ros/ros_comm #2279: Epydoc not available on Focal -> missing Python docs for many Noetic core packages

peci1, 2022-09-20:
Noetic doc jobs based on epydoc fail. E.g. https://build.ros.org/view/Ndoc/job/Ndoc__ros_comm__ubuntu_focal_amd64/50/console#console-section-7 . The docs are not generated at all: http://docs.ros.org/en/noetic/api/rospy/html/ (compare with [Melodic docs](http://docs.ros.org/en/melodic/api/rospy/html/) ).

The most notabl package with this deficiency is rospy!

Is there any contingency plan, i.e. some magic in rosdoc_lite which would automagically switch to sphinx on Noetic?

mikepurvis, 2022-10-10:
At Clearpath, we wrote and use https://github.com/mikepurvis/catkin_tools_document to generate our internal API docs— there was a discussion back in 2019 on that project about what to do with epydoc, see: https://github.com/mikepurvis/catkin_tools_document/pull/10#discussion_r341732792

Switching to Sphinx is non-trivial since it's conceptually so different from the more auto-documenting approach of something like epydoc or doxygen. We ended up deciding to go with pydoctor, so that's what catkin_tools_document does when it encounters a call for epydoc in the `rosdoc.yaml`. Unfortunately I don't have a live documentation page to link you, but it's also not hard to generate a few yourself if you want to check it out.

peci1, 2022-10-12:
@mikepurvis Great you've managed to get pydoctor working, it seems a reasonable substitution for epydoc. I see catkin_tools_document is a catkin_tools-only tool, so I guees it would not be possible to easily use it in the ROS Buildfarm. Or is there a way?