ros/ros_comm #2289: Kernel release parsing

MatthijsBurgh, 2022-10-10:
I am running Ubuntu 20.04 with the 6.0 kernel, installed via `ppa:tuxinvader/lts-mainline`.

I am not sure there are any rules that apply to the release name of a kernel. But it is `6.0-060000-generic`. While for 5.15 is `5.15.0-48-generic`. So the big difference here is that the patch version is not provided.

In rosgraph.xmlrpc you split the release on dots. So normally this would caused the last element to contain the patch version and all other release info. Now this is included in the minor version, which causes the current error.

```py
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/roslaunch", line 34, in <module>
    import roslaunch
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/__init__.py", line 57, in <module>
    from .launch import ROSLaunchRunner
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/launch.py", line 55, in <module>
    from roslaunch.nodeprocess import create_master_process, create_node_process, DEFAULT_TIMEOUT_SIGINT, DEFAULT_TIMEOUT_SIGTERM
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/nodeprocess.py", line 53, in <module>
    from rosmaster.master_api import NUM_WORKERS
  File "/opt/ros/noetic/lib/python3/dist-packages/rosmaster/__init__.py", line 35, in <module>
    from .main import rosmaster_main
  File "/opt/ros/noetic/lib/python3/dist-packages/rosmaster/main.py", line 43, in <module>
    import rosmaster.master
  File "/opt/ros/noetic/lib/python3/dist-packages/rosmaster/master.py", line 45, in <module>
    import rosgraph.xmlrpc
  File "/opt/ros/noetic/lib/python3/dist-packages/rosgraph/xmlrpc.py", line 103, in <module>
    class SilenceableXMLRPCRequestHandler(SimpleXMLRPCRequestHandler):
  File "/opt/ros/noetic/lib/python3/dist-packages/rosgraph/xmlrpc.py", line 104, in SilenceableXMLRPCRequestHandler
    if _support_http_1_1():
  File "/opt/ros/noetic/lib/python3/dist-packages/rosgraph/xmlrpc.py", line 94, in _support_http_1_1
    platform_minor = int(release[1])
ValueError: invalid literal for int() with base 10: '0-060000-generic'
```
https://github.com/ros/ros_comm/blob/f5fa3a168760d62e9693f10dcb9adfffc6132d22/tools/rosgraph/src/rosgraph/xmlrpc.py#L94

I suggest to change https://github.com/ros/ros_comm/blob/f5fa3a168760d62e9693f10dcb9adfffc6132d22/tools/rosgraph/src/rosgraph/xmlrpc.py#L92
to the following:
```py
 release = platform.release().split('-')[0].split('.')
```

peci1, 2023-02-22:
https://github.com/ros/ros_comm/pull/2202 fixed this issue and a binary release came out today containing that fix.

MatthijsBurgh, 2023-02-22:
Fixed by #2202

closed 2023-02-22 by MatthijsBurgh