ros/ros_comm #2296: python3.11 and roslogging (rospy.init_node)

pepipho, 2022-11-08:
test.py
```
import rospy

print("initializing node")
rospy.init_node( 'listener' , anonymous = True )
print("done")
```
Expected behaviour: init ros node and print "Done".
Current behaviour: `python3.11 test.py` is stucked in init_node method on this part
`rosgraph/roslogging.py`
```
class RospyLogger(logging.getLoggerClass()):
    def findCaller(self, *args, **kwargs):
        """
        Find the stack frame of the caller so that we can note the source
        file name, line number, and function name with class name if possible.
        """
        file_name, lineno, func_name = super(RospyLogger, self).findCaller(*args, **kwargs)[:3]
        file_name = os.path.normcase(file_name)

        f = inspect.currentframe()
        if f is not None:
            f = f.f_back
        while hasattr(f, "f_code"):
            # Search for the right frame using the data already found by parent class.
            co = f.f_code
            filename = os.path.normcase(co.co_filename)
            if filename == file_name and f.f_lineno == lineno and co.co_name == func_name:
                break
            if f.f_back:
                f = f.f_back
```
```
pip show rospy
Name: rospy
Location: /opt/ros/noetic/lib/python3/dist-packages
Version: 1.15.11
```

``` 
lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.2 LTS
Release:        20.04
Codename:       focal
```

destroy314, 2023-03-22:
Same question. For me, I just add `break` on the end of `while hasattr(f, "f_code"):`

jspricke, 2023-03-22:
See https://github.com/ros/ros_comm/pull/2297/commits/685a96ec9cd67f1fd6f8cd52cce6f251f8899e67

meijie-jesse, 2025-04-24:
> Same question. For me, I just add `break` on the end of `while hasattr(f, "f_code"):`同样的问题。我只需要在 `while hasattr(f, "f_code"):` 的末尾添加 `break` 即可：

its amazing magic!