ros/ros_comm #2323: Possibility to limit rosbag open memory requirements

kis-balazs, 2023-04-04:
Hi!

We are looking into extracting .csv files from ROS bags, and some of the bags are quite large (>40Gb). We tried limiting the soft limit of the memory usage for the python process executing - not to exceed certain required thresholds. This triggers a very specific MemoryException piping into "rosbag.bag.ROSBagUnindexedException: Unindexed bag" exception.

Do you have any recommendations how to limit/handle this in a better way? :)

Thank you!

kis-balazs, 2023-04-13:
Hi, any update on this? :)

mikepurvis, 2023-04-13:
@kis-balazs Questions of a "how do I" nature are better suited for answers.ros.org— this is a bug/feature tracker.

That said, I would try to implement your csv-extractor in C++ instead, as I believe the C++ interface to rosbag is more disciplined about not trying to load the whole thing into memory: http://wiki.ros.org/rosbag/Code%20API#C.2B-.2B-_API

closed 2023-04-13 by mikepurvis