ros/ros_comm #2390: XmlRpcValue - accessing a non-existing key now throws

zarmomin, 2025-05-13:
One side-effect of [this change](https://github.com/ros/ros_comm/pull/2315) is that retrieving a value of a non-existing key now throws an out of range instead of returning a default value.

sloretz, 2025-05-16:
Thank you for the report and minimal example @zarmomin !

I'm trying to figure out if this behavior change should be treated like a regression and reverted. Would you mind explaining how you discovered this?

mgrrx, 2025-05-17:
@sloretz this issue breaks a whole bunch of tests and packages on our side. Can you please revert the change until a proper fix has been provided?

zarmomin, 2025-05-19:
@sloretz Thanks for looking into this Shane! I noticed one of our unit-tests failing because of this - we were trying to access a non-existing value in a nested parameter. For us this actually caught a bug in the test, but it is unexpected to have this throw  change from `&` to `const&` and start throwing exceptions without any outward communication about its new behavior.

sloretz, 2025-05-19:
I had to make sure I understand what's happening. I'm going to revert #2315 because it seems like a significant behavior change. 

The new behavior introduced by #2315 seems better to me, but it should be a breaking change in a new ROS 1 distro, and there isn't going to be a new one of those.

---

The `XmlRpcValue::operator[]()` accesses `_value`, which in this case is of type `ValueStruct`, which is an `std::map<>` https://github.com/ros/ros_comm/blob/b6c57e76a764252cf50d8d24053f32e2ad54a264/utilities/xmlrpcpp/include/xmlrpcpp/XmlRpcValue.h#L43

## Before #2315
It looks like ros/ros_comm at 4755096d2675fadc61d91d029543739d45b029ac (Just before #2315) uses the `T& std::map<>::operator[]( const Key& key );` constructor, which inserts a default constructed value when the key doesn't exist.

> Inserts `value_type(key, T())` if the key does not exist.

## After #2315
At b6c57e76a764252cf50d8d24053f32e2ad54a264 (1.17.3, current `noetic-devel`) it uses `T& std::map<>::at( const Key& key );`. This throws an exception if the key doesn't exist

> [std::out_of_range](https://en.cppreference.com/w/cpp/error/out_of_range) if the container does not have an element with the specified `key`.

## Attempted fix #2391 

With my fix the code uses `T& std::map<>::operator[]( const Key& key );` again. However, `XmlRpcValue::operator[]()` returns type `XmlRpcValue const &` which in the given MRE is then cast to `const std::string&()`. This invokes the [`operator const std::string&()`  user-defined conversion function](https://github.com/ros/ros_comm/blob/b6c57e76a764252cf50d8d24053f32e2ad54a264/utilities/xmlrpcpp/include/xmlrpcpp/XmlRpcValue.h#L101) which raises an XMLRpcError. I think it's raising an error because [a default constructed XmlRpcValue ](https://github.com/ros/ros_comm/blob/b6c57e76a764252cf50d8d24053f32e2ad54a264/utilities/xmlrpcpp/include/xmlrpcpp/XmlRpcValue.h#L48) has its `_type` set to `TypeInvalid`, and [the const version of `assertTypeOrInvalid()`](https://github.com/ros/ros_comm/blob/b6c57e76a764252cf50d8d24053f32e2ad54a264/utilities/xmlrpcpp/src/XmlRpcValue.cpp#L75-L79) always raises if the type is not set. The original code before #2315 doesn't raise `XmlRpcError` because [the non-const version of `assertTypeOrInvalid()`](https://github.com/ros/ros_comm/blob/b6c57e76a764252cf50d8d24053f32e2ad54a264/utilities/xmlrpcpp/src/XmlRpcValue.cpp#L83-L85) sets `_type` and populates `_value` when `_type` is `TypeInvalid`.

* https://en.cppreference.com/w/cpp/container/map/operator_at
* https://en.cppreference.com/w/cpp/container/map/at
* https://en.cppreference.com/w/cpp/language/cast_operator

closed 2025-05-19 by sloretz

sloretz, 2025-05-19:
Released as 1.17.4 https://github.com/ros/rosdistro/pull/45848

@mgrrx @zarmomin Please try out the new release from the testing apt repos and report if it fixes the broken unit tests.

PR #2391 (2025-05-19): Revert: Maintain constness of accessed XmlRpcValue struct
Edit: This now reverts #2315  @rhaschke FYI

Reverting because it changed behavior in a way that broke user's builds. See #2390 for more information.

Fixes #2390